gdombiak/OctoPod #235: Push notification adjustment

gryn, 2019-12-26:
Thank you for considering the following feedback.

I would like to see the push notification customized under IOS10. The current layout is I believe a default, where the thumbnail ends up being about 1cm square. I think it would be a big quality improvement if this was attached below the text message and at full device width. Here is an article about the code changes needed:

https://developer.apple.com/documentation/usernotificationsui/customizing_the_appearance_of_notifications

Often I get these notifications when I am not at the house, so I cannot connect to the printer to get a preview. Having this thumbnail being larger would be a great help!

As a separate feature request, it would be nice to have a sort of egg-timer setting where this sort of notification would be sent periodically during the print. (Again to help with the use case that I'd like to see what is going on when I am away and unable to directly connect).

Finally, I'd like to thank you all for your hard work and the excellent job you have done with this application. Thank you!

jollino, 2019-12-27:
Not sure what iOS version you are on, but at least on 13.x I can long-tap the notification and the image shows up bigger: 
[image]

The egg timer idea is nice, however without being able to send commands (I.e. when on a different network) that would essentially be read-only. I’m not sure if allowing bidirectional communication to stop the print in case of trouble is even possible without some sort of intermediate node. I personally use a telegram bot via an octoprint plugin to both get images every N minutes and send commands if needed, with telegram being the intermediate node itself as it can be accessed from anywhere (i hope that makes sense!)

gryn, 2019-12-28:
Jollino. Thank you very much for this information and the reply. That does indeed work.

I'll look up the telegram app. I realized that the egg-timer idea would be one directional. But, I could always call the fire department if my house was already on fire :) .