gdombiak/OctoPod #237: The Files view shows only source "SD card or "Octoprint", but no file names.

IonicEV, 2020-01-08:
I am on octopi 0.14.0 running OctoPrint 1.3.4. If seems that the file names are missing in the file listing. and when you select specific file - it shows all info (size, origin, print time, etc.), but File name. When you choose "Sort by" as "Printed" - the all crashes and there is no way to get back to Files view as it would crash app with saved sort by order. The only way to get Files back is to  reinstall the app.

gdombiak, 2020-01-08:
Thanks so much @IonicEV for reporting this issue. It helps a lot. Could you also share a screenshot of the previous screen that lists the files? I will install that OctoPrint version and try to reproduce it.

Gaston

IonicEV, 2020-01-08:
Looks like I dropped the same screenshot originally. I added more screenshots.

gdombiak, 2020-01-08:
Good news. I was able to reproduce this error when using OctoPrint 1.3.4. I see that it was released on Jun 1, 2017 so that is 1 year older than OctoPod. I will see what I can do to prevent this app crash in the next release.

In the meantime, do you think you can update OctoPrint to use a newer version? 1.3.9 or newer will work with your version of the app. OctoPrint has many security fixes and also improvements since your version.

My best recommendation for security reasons would be to:
1. Make a backup of OctoPrint and download the backup file to your computer
1. Download and burn the latest OctoPi. Your version (unless you manually kept it up to date) is very old and Raspbian may have many many security vulnerabilities
1. Once OctoPrint is up and running then restore the backup. Whatever plugins you have will be installed again and their configuration will be restored

I will update this thread with my attempt to make OctoPod run with this old version of OctoPrint.

IonicEV, 2020-01-08:
There is a reason folks are on 1.3.4 - it is very popular M33-Fio plugin to support M3D micro with iMe firmware. The plugin does not work with higher versions. Also the integrated Cura slicer is another reason to stay on this version. The missing file name is not bothering me, but printer sort order is a bit hard to recover from.

gdombiak, 2020-01-08:
No worries. We will get this sorted out. :) Working on this now.

gdombiak, 2020-01-08:
It's fixed now. Old version of OctoPrint does not have _display_ information but just _name_ so I'm now using name when there is no display information. I have it working now! Will push new version to the app store tonight and maybe it is available in the app store tomorrow.

Gaston

IonicEV, 2020-01-08:
Many thanks! I really like Octopod out of other apps.

closed 2020-01-08 by gdombiak

gdombiak, 2020-01-08:
FYI, app submitted. Apple needs to review and approve. Should be love tomorrow!

gdombiak, 2020-01-08:
New version is now up in the App Store. Let me know how things work for you.

Thanks again for reporting this,
Gaston

IonicEV, 2020-01-08:
Thanks! The app is working, I did not have a chance to check on printed sort by. Will do it  later when octopi is accessible,

commit 88b1ba9e (2020-01-08): Fallback to use 'name' when there is no 'display' field information. Needed for old versions of OctoPrint that has an old API.

Closes #237