gdombiak/OctoPod #248: Printer is restarting on Connect

Ruko2010, 2020-01-19:
Every Time when I first connect the app to my printer the printersoftware is restarting and of cause failing my print. 
Even when I connected octoprint via webinterface first and then connect the App the printersoftware is restarting. 
Maybe #170  this is the same problem. Hope somebody can help me.

gdombiak, 2020-01-19:
Hey @Ruko2010 ,

Thanks for reporting this. Is the printer being reset when you click _Connect_ ? If that is what you are reporting then that is normal with most 8-bit boards. For instance, my Prusa MK3 does when there is a new USB connection. My Ender 3 used to do this when it had the 8-bit board. Once I upgraded the board to a 32-bit it no longer gets reset/reboot.

Is this what you are seeing?

Thanks,
Gaston

Ruko2010, 2020-01-19:
Yes, this is what I see. Most of the time my workflow is to connect the printer via webinterface and then again via the app (2 reboots) and then I start my print. But sometimes I forget the app connect and then I can‘t use the app. This was a bit annoying and I thought maybe I just configured something wrong. 
But if this is normal, then I have to deal with it 😃

jollino, 2020-01-19:
That is actually weird. The reboot when you first connect in the web interface is normal, but at that point OctoPod should NOT require another connect. I always connect to the printer from a browser on my computer as I mostly use OctoPod to monitor the print, and I find it already connected.

gdombiak, 2020-01-19:
As Daniele said, once OctoPrint connected to the printer then it should appear as Connected in the app and the web interface. Things should be in sync. I wonder if OctoPod is having an error connecting to OctoPrint via websockets. Do you see an error like "Not Refreshing" in OctoPod after a while?

jollino, 2020-01-19:
By the way (slightly OT) I'm trying to replicate an issue I'm having with the current stable version: when wi-fi acts up or I switch networks, sometimes OctoPod doesn't notice and I have to force quit and reopen it. I was thinking that maybe it could do some sort of reachability test to the server's IP within the first few seconds of being open/awake, and if that fails, re-establish the websocket connection. As soon as I have some more solid data to report on, I'll open a specific issue.

Ruko2010, 2020-01-19:
> As Daniele said, once OctoPrint connected to the printer then it should appear as Connected in the app and the web interface. Things should be in sync. I wonder if OctoPod is having an error connecting to OctoPrint via websockets. Do you see an error like "Not Refreshing" in OctoPod after a while?

 I‘ll check that the next time. But I think OctoPod is not showing a connection to the printer (no status), so most of the time I close the App and lunch it again.
I‘ll watch my steps next time I fire up my printer. 
(I‘ll reopen the issue until further testst)

gdombiak, 2020-01-19:
@Ruko2010 sounds good. If there is no temp info or other things on the main screen then wait a little bit. If you see a "Not refreshing" error message then press it and it will tell you what the problem is. With that info we can troubleshoot the connectivity problem.

@jollino OctoPod has recovery logic when connection is lost. I can turn on-off wifi and it switches to cellular or vice versa. I wonder if in your case the iOS is not realizing that things went south. OctoPod has a silent heartbeat every 30 seconds (IIRC) so if that does not fail then we cannot do much about it. The heartbeat runs on the websocket itself.

jollino, 2020-01-19:
> @jollino OctoPod has recovery logic when connection is lost. I can turn on-off wifi and it switches to cellular or vice versa. I wonder if in your case the iOS is not realizing that things went [...]

You're probably right, because other apps seem to choke too when that kind of thing happens. I'll use percussive methods on the access point. ;-)

Ruko2010, 2020-01-26:
Ok, I‘ve tested it again and I think I was not patiened enough. If I connect the printer via Laptop and then just open the app and wait it does not restart. It just connects. Maybe I hit „connect“ to early. Thanks for the help..

closed 2020-01-26 by Ruko2010