gdombiak/OctoPod #251: Add info in full screen webcam view

nexux78, 2020-01-21:
Is possible to add some information in full screen mode as layer height, bed and hotend temp, etc?

gdombiak, 2020-01-21:
Great idea @nexux78. I can add bed and hotend temps. Marlin does not send OctoPrint any information about current Z or layer height. This means that OctoPod cannot have it either.

Perfect timing for this issue. I scheduled it for the next major release that should be out next week.

gdombiak, 2020-01-21:
I will see if I can add also ETA

closed 2020-01-23 by gdombiak

gdombiak, 2020-01-28:
[image]
[image]

commit cfe0279f (2020-01-23): Fixed and tested more display printer status info on the webcam view

Closes #251