gdombiak/OctoPod #254: Disconnect button should require confirmation

accountnumber3, 2020-01-26:
I just cancelled a print because I wasn't connected to WiFi when I opened the app. I joined WiFi in the process but the app did not connect to octoprint immediately. 

I clicked connect.
The button changed to disconnect. 
I got a notification from my telegram bot that the print failed. 

Regardless of having connect and disconnect use the same space, the real issue is that the disconnect command is issued to octoprint, and not to the app. If that is the logic, then disconnect needs a confirmation prompt.

jollino, 2020-01-26:
I'm not currently printing, but I gave it a try and I do get a confirmation prompt.
[image]

gdombiak, 2020-01-26:
Hey @accountnumber3,

As Daniele said, OctoPod has a confirmation dialog on connect and disconnect. You can enable/disable this confirmation from the Settings tab.

[image]

[image]

accountnumber3, 2020-01-27:
Ok, I see it too. I think what happened is I was in a hurry and was tapping the connect button because I thought it meant connect for octopod to octoprint. I essentially created a race condition where I told octoprint to connect to the printer even though it was already connected. 
I guess I'd like a feature request to prevent commands from being sent if the state isn't verified.

gdombiak, 2020-01-28:
Scheduled for the next release. Will check that the connect/disconnect button is active only after we really know the state.

accountnumber3, 2020-02-01:
I see some really great updates. Not sure if this made it in but I'm not gonna test it. You can close this if you want

gdombiak, 2020-02-01:
Still need to check that the connect/disconnect button is active only after we really know the state. Trying to implement for the coming release

commit 4ad49905 (2020-02-02): Disable connect/disconnect button until we are certain about connection state from OctoPrint to the printer.

Closes #254
closed 2020-02-02 by gdombiak