gdombiak/OctoPod #255: Custom Controls not available during Print

Ruko2010, 2020-01-28:
When I start a print, I cannot use my custom controls in octoprint to enable/disable my LED Strip. I installed the Strip to light up the printing area so I can see what is printing via webcam when its dark in the room. 
The strange thing is, that I can use the controls via the octoprint webinterface, even during print. The controls are working when no print is in progress. 
[image]

gdombiak, 2020-01-28:
Hi @Ruko2010,

OctoPod has a safety feature that when enabled it will disable certain actions while printing to not interfere with the print job.  Custom Controls can potentially interfere with the print so they get disabled when this safety feature is enabled.

Go to Settings -> Security and disable "Auto-lock while printing". You can also manually disable "Lock read-only mode" and Custom Controls will become enabled again.

gdombiak, 2020-01-28:
BTW, update to the recently released OctoPod 3.0 and enjoy the many new features. :) It went live today a few hours ago.

Ruko2010, 2020-01-28:
Thanks for the really quick help. I‘ll test it on my next print. And yes, I‘ve already updated 😉. 
Btw, there are some german translations that won’t sound very good. Do you need help with that?

gdombiak, 2020-01-28:
Sounds great.

Sure thing. You can see the translation in context which helps a lot. I try to provide as much context to translators as possible but things might get lost "in translation". :) Let me know which things I should update and I can update them for the next release.

Thanks,
Gaston

Ruko2010, 2020-01-29:
I turned the safe option off and now the Custom control works while printing. Thanks for the help.

> Btw, there are some german translations that won’t sound very good. Do you need help with that?
I re-read the translations and realisied, they sound good. Maybe it was too late last time...^^

closed 2020-01-29 by Ruko2010