gdombiak/OctoPod #256: Terminal view improvements

jollino, 2020-01-28:
When using the landscape view, landscape mode is great to see full lines, but it also means seeing fewer of them.

Compare:
[image]

and:
[image]

The keyboard makes things worse:
[image]

It would be nice if the app could "make room" to see more text, e.g. by hiding the tab bar below.
Even the update switch could be moved to the settings or removed entirely, in my humble opinion: I can't think of a single use case where you'd be in the terminal view without the terminal itself being updated...

gdombiak, 2020-01-28:
Thanks Daniele for more great feedback. I do use the stop refreshing feature. :) What do you think if I switched the orders of those 2 fields. Comando GCode will appear first so that will give more room for the drop down. I will post a screenshot with that change before deciding adding logic to hide the numbers bar. Sounds good?

jollino, 2020-01-28:
I admit I never tried disabling it, I'll have to see what it does. :D 
That would definitely help with the dropdown, in fact I was thinking about maybe having a separate button for the dropdown (maybe to the very right of the field, especially if the Send button is removed in favor of just pressing enter on the keyboard): tapping that button would show/hide the dropdown and, if needed, hide the keyboard.
It would also have the side effect of being able to type in the field freely without having the dropdown hide anything, even in portrait mode.
Not sure if that makes sense though, it's probably a lot of work for a very narrow use case.

On a side note, I noticed a few little things about the Italian translation that could be improved; I'll message you on FB with them. Did you happen to try any of those localization platforms, by the way?

gdombiak, 2020-01-28:
Thanks so much Daniele. I haven't had time to check out those localizations tools. :( Will do as time allows.

Thanks,
Gaston

commit f65766ab (2020-02-02): Reduced number of visible rows on commands history in terminal window

Closes #256
closed 2020-02-02 by gdombiak