gdombiak/OctoPod #257: Watch app: pull to refresh

jollino, 2020-01-28:
A simple suggestion: pull to refresh is a common idiom on the Watch as well (Mail uses it, for instance) so it could be used in Octopod as well. I'm often terrified of accidentally hitting pause or stop (does it ask for confirmation? I never tried) when reaching down for refresh!

[image]

gdombiak, 2020-01-28:
Any potentially destructive action comes with a confirmation dialog. [See for pause](https://github.com/gdombiak/OctoPod/blob/92c8098af4a0b5dc7ca46fe0e76c7cf253cab225/OctoPod%20Watch%20Extension/Panel/PanelInterfaceController.swift#L159) and [here for cancel](https://github.com/gdombiak/OctoPod/blob/92c8098af4a0b5dc7ca46fe0e76c7cf253cab225/OctoPod%20Watch%20Extension/Panel/PanelInterfaceController.swift#L181)

I like your idea of adding a pull down for refreshing the info. I can remove the refresh button from all pages if possible. Sounds good?

jollino, 2020-01-28:
That sounds perfect, I think it would be pretty obvious for most users.

A little unrelated but I also noticed — I had taken a bunch of screenshots the other day and I'm just going through them now — that the icon in the small complication seems a little "weird". It's hard to explain, but it almost looks like there's no anti-alias, and it's a little jarring compared to everything else:
[image]
I think it could be much more readable if the lines were thicker, as Apple Watch has a tendency to use strong sub-pixel smoothing at a hardware level. Right now they're so thin that the smoothing just looks odd.

(I really wish I could test these things myself, I should set up Xcode properly at some point. :D)