gdombiak/OctoPod #258: Issues with Kasa HS105

DaVinciQc, 2020-01-29:
Hi,
I have a TP-Link HS105 smart plug that I'm controlling either with the Kasa app on my Android or in the Octoprint browser interface via the TP-Link Smartplug plugin. Everything was working fine until Octopod automatically added the smart plug on my iPad interface (I didn't add it myself).

Issue #1 : The icon in Octopod (iPad) doesn't show the right power state of the plug. When the plug is ON (physically and on Octoprint interface) Octopod shows an icon of a 2-prung plug with a red X over showing the plug is OFF. And vice-versa. When the plug if OFF Octopod shows the plug icon WITHOUT the red X. The icon must indicate the actual power state of the plug. And now Octopod is doing the opposite. I've reboot both Octoprint and Ocopod and it changed nothing.

Issue #2 : The plug gets automatically turn OFF. The issue occurs when in TP-Link Smartplug Editor both "Enable error event monitoring." and "Off on Error" are enable (box checked). Each time in Octoprint I turn the plug ON, a few seconds the plug is automatically turn OFF. Then I get two notifications : 

In Octoprint :  "State: Error: Failed to autodetect serial port, please set it manually."

In Octopod a similar Error:Failed notification appears on my iPad. 

If I disable the "Off on Error" option everything goes back to normal. 

I'm guessing it's Octopod who sends an error signal to Octoprint, then the TP-Link Smartplug plugin intercepte this error and shuts down the plug.

I would like to delete the plug from Octopod but I didn't find how.

Thank you for your help.

gdombiak, 2020-02-01:
Hi @DaVinciQc,

Sorry for the issue you are running into.  The icon that you see in issue 1  indicates the action that will happen if you click on the button. It does not show the current state of the plug.

OctoPod uses the API that is exposed by the TPLink plugin. Clicking on the image/button to turn the plugin on/off has the same effect as clicking on the OctoPrint web UI. Both functionalities use the same API.

Not sure if my answers help. :(
Gaston

DaVinciQc, 2020-02-01:
Hi,

1) I was afraid you were gonna give me this explanation. It's wrong. That's not how things normally work in an interface. The icon MUST show the status and NOT the action that will be happen. Years of interface development has proven that already. Let's take iOs as an exemple since I personally admire iOs interface for years. On an iPhone / iPad we have multiple toggle switches. When a toggle is green it means that feature is turn ON, not that it would be turn ON if we switch the toggle from green to clear.

Please don't be offended. You have a good product here. You have done good work on Octopod and I'm happy I can use it for free. During my years of experience as senior IT project management for our main financial institution here I have encountered this exact problem on a major HR software suite we were implementing at our head office. In pre-prod version some icons were showing the future state (action that will happen) if the user was to click on them instead of showing the actual state. As the client representative I urgently ordered the developer to change that before going to prod because it was making things extremely confusing for our test users, including myself. The way the plug icon is designed in Octopod goes against all good practice in the industry.

2) You're telling me it's impossible that the Octopod app is the culprit here, impossible that Octopod is the one sending an error to Octoprint?

Regards,
Mike

gdombiak, 2020-02-02:
Hi Mike,

I'm happy to make changes to OctoPod to keep improving it. As you can tell, I'm not a UI developer so not super familiarized with best UI practices. I'm open to all feedback to improve the app. :)

I see your point and now that I think about it, it is true that iOS shows current state and not the action. Toggles come to mind as an example.  Easy change to flip icons.

I'm not saying that it is impossible for OctoPod to have an error. :) Perfection is an illusion. I'm describing how things work by design. It would help if you could intercept traffic towards your OctoPrint instance to confirm that things work as designed. Would that be possible?

Thanks,
Gaston

gdombiak, 2020-02-02:
Fix committed. Icons now indicate current plug state.

DaVinciQc, 2020-02-02:
Wow great! Thanks I'll try that tomorrow.
P.S. Does the iOs version of Octopod updates automatically through the App store?

gdombiak, 2020-02-02:
Hey Mike,

I submitted the first beta for the next release 3.1 to the App Store. Once Apple reviewed and approved you can give it a try. It has many other nice improvements as well. Send me a message to gdombiak at gmail if you are not part of the beta testing team and would like to join.

Should we close the ticket or is there anything else?

Thanks,
Gaston

DaVinciQc, 2020-02-02:
Don't close the ticket yet. You didn't tell me how to monitor the traffic toward Octoprint to help determine if Octopod is the one sending error to Octoprint regarding my Tp-link plug that's being turn off automatically after few seconds.

Téléchargez Outlook pour Android<https://aka.ms/ghei36>

gdombiak, 2020-02-02:
The easiest way to monitor OctoPrint's traffic is to set logging if tornado.access to info. See screenshot below:

[image]

Web traffic will be logged to octoprint.log like this

[image]

gdombiak, 2020-02-11:
Mike,

I'm wrapping up things to release 3.1 Can I close this ticket?

Thanks,
Gaston

gdombiak, 2020-02-15:
Mike, I will close this ticket for now. We can reopen it or open a new one if you have new info about what you were seeing.

Thanks,
Gaston

DaVinciQc, 2020-02-16:
But Gaston you never answered my email of February 6th. Could you please answer my email before closing the ticket?

Téléchargez Outlook pour Android<https://aka.ms/ghei36>

Closed #258<https://github.com/gdombiak/OctoPod/issues/258>.

gdombiak, 2020-02-16:
Hmm, not sure which email you are referring to. :(  Could you resend it or post it here?

closed 2020-04-17 by gdombiak