gdombiak/OctoPod #260: Webcam stream and Reverse Proxy

philgood, 2020-02-01:
Hi Gaston,

Firstly I really enjoy OctoPod and use it regularly. The new UI refresh is excellent too. 

Before raising an issue, I have been through the FAQs and current open/closed issues looking for the problem I am having. I couldn't see anything related. If I missed it, I'll close this issue. 

I have just configured a reverse proxy so I can access services such as OctoPrint remotely for convenience. After setting up OctoPrint WebUI to handle the reverse proxy subdomain. I changed the Stream URL to '/octoprint/webcam/?action=stream' this is to allow for the webcam subdomain redirection. Everything works as expected for the OctoPrint WebUI.

I configured OctoPod to access OctoPrint, it connects fine and I have control of the printer. The only thing that doesn't function is the webcam stream. I get a 'Request error. HTTP response: 404'

I also notice that the webcam stream url is 'https://domainname.org/octoprint/octoprint/webcam/?action=stream'.  I feel it should be 'https://domainname.org/octoprint/webcam/?action=stream' which I tested this url in a browser and the webcam stream loads.

[image]

When I reverted the Stream URL back to '/webcam/?action=stream' on OctoPrint the webcam stream works as expected on OctoPod, but doesn't on OctoPrint WebUI. 

I did read ticket https://github.com/gdombiak/OctoPod/issues/61 about not hard coding and using the api/settings. I think in this case hard coding would work better for me. 

Do you have any possible solutions with how the api settings are read? Is there anything I'm missing or could try and/or change?

gdombiak, 2020-02-01:
Thanks Phil for the detailed report. That double octoprint in the path is certainly suspicious. As you may know, when the MultiCam plugin is not installed (your case I think) then the URL to the webcam is treated like this in OctoPod:

> camera URL = [The URL that you specified for your printer in OctoPod] + [cameraPath]

where cameraPath is what you defined in _.octoprint/config.yaml_ under _webcam -> stream_. If that setting does not exist then it will default to _/webcam/?action=stream_.

So let's say that your printer URL in OctoPod is **https://domainname.org/octoprint/** and your config.yaml has **/octoprint/webcam/?action=stream**, then the camera URL will end up **https://domainname.org/octoprint/octoprint/webcam/?action=stream**. I think this is your case.

Having said all this, I'm not clear how you are using the path octoprint in your setup. Is it only something that exists in your reverse proxy? If yes then config.yaml should have _webcam/?action=stream_ in webcam->stream and then in your reverse proxy you will need to redirect to the proper URI. For instance, if your public URL is _https://domainname.org/octoprint_ and the internal hostname of your OctoPrint instance is _octopi.local_ then you can have a forward to the internal hostname. If you are using haproxy then look at the acl command. 

Hope this makes sense.
Gaston

philgood, 2020-02-02:
Hi Gaston,

Thanks for getting back to my inquiry so quickly.

I fully understand what you are saying and I was definitely on the right path. I've found through forums most people who do a reverse proxy for OctoPrint sit it on the root name space. Because I made a sub path for OctoPrint eg https://domainname.org/octoprint this made things tricky with adding the webcam as a further sub path. 

After loosing another day playing/learning HAProxy I found out by putting the whole name space in the config.yaml (https://domainname.org/octoprint/webcam/?action=stream) worked. I was blown away!

These are the changes I made to /etc/haproxy/haproxy.cfg

```
frontend public
        bind :::80 v4v6
        bind :::443 v4v6 ssl crt /etc/ssl/snakeoil.pem
        option forwardfor except 127.0.0.1
        use_backend webcam if { path_beg /octoprint/webcam }
        default_backend octoprint

backend octoprint
        acl needs_scheme req.hdr_cnt(X-Scheme) eq 0
        reqrep ^([^\ :]*)\ /octoprint/(.*) \1\ /\2
        reqadd X-Script-Name:\ /octoprint
        reqadd X-Scheme:\ https if needs_scheme { ssl_fc }
        reqadd X-Scheme:\ http if needs_scheme !{ ssl_fc }
        option forwardfor
        server octoprint1 127.0.0.1:5000
        errorfile 503 /etc/haproxy/errors/503-no-octoprint.http

backend webcam
        reqrep ^([^\ :]*)\ /octoprint/webcam/(.*)   \1\ /\2
        server webcam1  127.0.0.1:8080
        errorfile 503 /etc/haproxy/errors/503-no-webcam.http
```

I wanted to explain what I did so if anyone else comes across the issues I had, this can hopefully help them out.

gdombiak, 2020-02-02:
Hey Phil,

Great to hear that you figured it out. Thanks so much for sharing your setup so others can use it in the future.

Can we close this ticket now?

philgood, 2020-02-02:
Sure can and thanks again!

closed 2020-02-02 by philgood