gdombiak/OctoPod #261: Feature request: UX improvements

accountnumber3, 2020-02-01:
After finishing a print I wanted to start another. My printer is not in the house so I figured I could use the webcam and controls nudge the print off the bed. It would have worked except that I'm stupid and impatient in all the worst ways. The part that I had printed is short but wide. I moved the Y axis as far out as it could go and X was somewhere in the middle. Being a complete moron I decided that the fastest way to get below the top of the print was to home Z. 
My maker select's cooling fan has a really solid steel mount which blended into the background and I forgot existed. I just turned that steel mount into a hydraulic press and smashed the shit out of my print. Except the print is nearly solid as well. The part that isn't so solid is the X carriage bars. I haven't gone out to check but it doesn't look good. In my panic the only things I could do was take a screenshot, and I didn't even do that.
Much too late I disconnected from the printer, reconnected, and sent Z shooting straight up. Octopod was set to XY increments of 10 while Z was 100. I probably clicked it a dozen times. 

Safety feature requests:
1. Disconnect button should be on every tab. It took a while to find it.
2. Tactile feedback on movement commands. I couldn't tell if I was actually hitting the buttons. Strength and/or duration should be variable based on the movement increment setting. Increments of 100 should be heavy and long. Increments of 0.1 should be light and short.
3. Add an `A` option to disable all motors, like a full stop.

gdombiak, 2020-02-01:
Excellent excellent feedback! Implemented and now testing your great idea 2. Will add 3 later. Will see if I can do 1 for the next release. That will require more work.

accountnumber3, 2020-02-01:
I don't know iOS programming but I imagine you could easily add a standard short vibration to all commands and make it fancy later.

For the disable all motors thing, I mean on the Move tab under Fan & Motors > disable motor, you already have `X Y Z E`. I'm suggesting to add `A` to that list to do all four. Just in case that wasn't clear.

Feel free to split these into multiple requests if you want

accountnumber3, 2020-02-01:
Also, good news my printer is ok. Just had to re-level it. 😅

gdombiak, 2020-02-01:
Haptic feedback tested and now committed. It's coming in the next release. Got to go now but will keep implementing your great ideas.

Happy to hear your printer survived! It was nice readying your story. Good writing style.

jollino, 2020-02-01:
> For the disable all motors thing, I mean on the Move tab under Fan & Motors > disable motor, you already have `X Y Z E`. I'm suggesting to add `A` to that list to do all four. Just in case that wasn't clear.

Maybe, for consistency with the axis homing buttons, we could have a general "disable all" and then one additional button for each axis?

gdombiak, 2020-02-01:
So add a new button to be able to "disable all" motors, correct? I think you both are suggesting this same thing.

accountnumber3, 2020-02-01:
I'm not sure what he's asking for.

I'm looking for a little `A` or `All` right here:

[image]

gdombiak, 2020-02-02:
Hey @accountnumber3, I think that is what Daniele is suggesting too. Due to localization needs, I'm leaning towards an icon rather than an A. IPhone SE/5/6 have narrow screens and some translations are large so no space for a whole word. :( 

How does any of these icon look for indicating multiple motors?
[image]

My vote is on the first icon.

Thanks
Gaston

gdombiak, 2020-02-02:
Disable all motors button has been implemented for next release.

[image]

commit b6a34ba0 (2020-02-02): Added new option to disable all motors at once

Closes #261
closed 2020-02-02 by gdombiak