gdombiak/OctoPod #264: Add info on number of Palette splices

MimbaMonkeyHouse, 2020-02-11:
It would be great to have information on the number of splices already done and the total number of splices for the current print directly from OctoPod.

gdombiak, 2020-02-11:
Hi @MimbaMonkeyHouse,

Maybe I'm misunderstanding the request, is the information you are requesting different from what appears in the second row of the Palette 2 section?

[image]

MimbaMonkeyHouse, 2020-02-15:
Hey @gdombiak. Sorry, I'm a monkey and missed this.
I'm trying to get info from Palette to use in Home Assitant. I can get the pings by using:
```
$.ajax({
                url: API_BASEURL + "plugin/palette2",
                type: "POST",
                dataType: "json",
                data: JSON.stringify({
                    command: "getPingHistory"
                }),
                contentType: "application/json; charset=UTF-8"
            }).done(function(data){console.log(data)});
```
This gives the entire history rather than just the last Ping that I need.
Can you let me know how you get the Splice (total and current), pings and pongs fro mthe Palette Plugin?

gdombiak, 2020-02-15:
Hey @MimbaMonkeyHouse (monkey) :)

OctoPod executes the command **uiUpdate** which is the same thing that the plugin does in the OctoPrint's UI. However, that HTTP response does not include the data. Instead the plugin will write back via the websocket and pass all the data that you see in the OctoPrint's UI. OctoPod then parses the JSON that came back in the websocket to extract the splices and the rest of the information.

I had to add **getPingHistory** for Siri support and some stats that I show in the UI and since I added that PR, I include the data in the HTTP response (unlike how the **uiUpdate** command works)

MimbaMonkeyHouse, 2020-02-15:
Thanks a lot for getting back to me on this. Nice to have some leads to follow... Wish you well with Octopod. Looks great. Too bad I have no iOS!

closed 2020-02-15 by MimbaMonkeyHouse