gdombiak/OctoPod #265: Request for Layer Number in Panel

Neiljclements, 2020-02-12:
Sorry if this is in the wrong section, however I can’t find the request for a feature.

I may have asked for this before, however would it be possible to have the current layer reported to the Panel page? This would be really useful when performing filament swaps for colour changes.

If it meant adding to the gcode file, like you can to report layers on the printer screen, that would still be very useful.

gdombiak, 2020-02-12:
Hi @Neiljclements,

I really wish we could show the current z layer but that is not possible at the moment. Marlin does not give that information to OctoPrint and therefore OctoPrint cannot give it to OctoPod. At some moment, OctoPrint used to calculate this information but since it was off then it got removed.

I can imagine that OctoPrint could offer this if you are printing from OctoPrint itself but it would still not be accurate since OctoPrint can ask Marlin to move to a given Z layer but it is up to Marlin's scheduler/queue to decide when that is going to happen.

TheLion, 2020-02-15:
Hi @gdombiak ,

Isn't it possible to rely the DisplayLayerProgress plugin to show layer information?

gdombiak, 2020-02-15:
Hi @TheLion,

That's an interesting idea that is probably good enough. That plugin reads the layer from the gcode that is being printed. It is not the exact layer the printer is in but it is probably enough to have an idea around which layer we are in. It will only work for slicers that have been configured to include this info in the gcode. 

That plugin reads the gcode being printed and displays the info on the screen. Will look for another plugin that also reads the gcode and instead writes the layer (not the LCD message) on the websocket so OctoPod can use it.

Thanks for the hint. Will look into adding this to the next release once I found a plugin that does this.

TheLion, 2020-02-15:
No problem. 

Octoprint-Dashboard (https://github.com/StefanCohen/OctoPrint-Dashboard) also uses the DisplayLayerProgress plugin to display layer information, so shouldn't OctoPod also be able to use that information?
(Excuse my ignorance, I have absolutely no idea what I'm talking about when it comes to programming or writing plugins ;-))

Neiljclements, 2020-02-15:
I’m liking where this is going 😊

gdombiak, 2020-02-15:
I'm thinking, in fact, of displaying layer info if you have any of these 2 plugins: [DisplayLayerProgress](https://plugins.octoprint.org/plugins/DisplayLayerProgress/) or [LayerDisplay](https://plugins.octoprint.org/plugins/layerdisplay/). Whatever you have, if the info is available then I can display it.

gdombiak, 2020-02-15:
Surprise ......

[image]

Neiljclements, 2020-02-15:
YOU BEAUTY!!! Oh man, I’ve wanted this for soooooo long.

Thank you so much Gaston!

Neiljclements, 2020-02-15:
Thanks for the input that got this off the ground @TheLion.

TheLion, 2020-02-15:
@Neiljclements NP. 😉

gdombiak, 2020-02-15:
While we are at it .....

[image]

Let's show more info exposed by the plugin. Adding **current height** if you are using https://plugins.octoprint.org/plugins/DisplayLayerProgress/

gdombiak, 2020-02-15:
Cannot make https://plugins.octoprint.org/plugins/layerdisplay/ work. Seems to be abandoned that plugin. Will only add support for https://plugins.octoprint.org/plugins/DisplayLayerProgress/.

commit 9414f835 (2020-02-15): Added "current height" and "layer information". Depends on DisplayLayerProgress plugin.

Closes #265
closed 2020-02-15 by gdombiak