gdombiak/OctoPod #266: Add Russian translation

Caribsky, 2020-02-16:


gdombiak, 2020-02-16:
Hi @Caribsky , this is super exciting!!! Would you be able to help adding a Russian translation? :D

Caribsky, 2020-02-16:
Hi @gdombiak 
Sure!

gdombiak, 2020-02-16:
AWESOME! Do you have a facebook account? We have FB group for translators that we use for communication and when new translations are needed. I attached the XLIFF file that needs translation. You will need to use some XLIFF editor. If you use a free one make sure that it lets you save the translated text. Some people did a full translation to discover they needed to start all over again.

[ru.xliff.zip](https://github.com/gdombiak/OctoPod/files/4209278/ru.xliff.zip)

 Thanks and welcome to the family! :)

Caribsky, 2020-02-16:
I wrote a message you on FB. 
Usually I use Xliff Editor (xliffedit.com)
Thanks

gdombiak, 2020-02-16:
That's perfect. I use the exact same tool. It's great!

Caribsky, 2020-02-18:
Hi @gdombiak! 
File with translation
[ru.xliff.zip](https://github.com/gdombiak/OctoPod/files/4217965/ru.xliff.zip)

Caribsky, 2020-02-20:
Updated translation.
Overall improvements for iPhone 5/5s/SE
[ru.xliff.zip](https://github.com/gdombiak/OctoPod/files/4230307/ru.xliff.zip)

gdombiak, 2020-02-21:
Thanks @Caribsky. I just incorporated your last version.

commit 2fdc5ecf (2020-02-21): Updated Russian translation

Closes #266
closed 2020-02-21 by gdombiak