gdombiak/OctoPod #267: [Feature Request] Persistent notification when printing

simpat1zq, 2020-02-18:
I don't know if this is possible, but is there any way to have a persistent notification that is always on the lock screen when the print is ongoing? Maybe with a little bit of progress information.

gdombiak, 2020-02-18:
Never saw a persistent notification. This is an interesting idea. Will check it out. Have you see the Today's widget? It shows progress and remaining time for all your printers in OctoPod.

simpat1zq, 2020-02-18:
Nice. Did not know about the widget. I just enabled it and put it at the top. I do feel that a persistent notification(again, if it's possible) would be a lot more useful since I the lock screen is a little more 'front and center' than the widgets screen.

simpat1zq, 2020-02-18:
Also, I just Googled it, and in the notifications settings, there is apparently in option to make notifications persistent, whereas temporary is the default. I turned it on just now, but I'm not 100% sure what that does.

gdombiak, 2020-02-18:
Hey @simpat1zq , been reading about persistent or sticky notifications and from what I see it is up to users to configure how they want to treat certain notifications (e.g. if they want to stay and not disappear from the UI). However, I cannot see that it is possible to refresh content of a notification that is already on the phone. I guess that is what you were after.

jollino, 2020-02-18:
I don't think persistent notifications are possible in iOS, not the way they exist in Android anyway (where some of them just get sticky and won't die). It is however possible for apps — well, for their back-ends anyway — to 'cancel' notifications already sent, so they disappear. iMessage is a prime example of that, where reading messages on a Mac will make the notifications disappear on the phone; and this is certainly available to other third-party apps too because Telegram does exactly the same thing. I'm not sure if it's possible to just nuke all previous notifications or if they have to be surgically removed one by one, but it may be possible to do so and send new ones, so that there's just one with the latest data. That's the closest to a persistent notification I can think of, personally.