gdombiak/OctoPod #271: Cameras on Apple Watch is empty

DeastinY, 2020-03-12:
Works fine within the app

gdombiak, 2020-03-15:
Hi @DeastinY, thanks for reporting this. Could you share your watchOS version, iOS version and Apple Watch model? Is there anything else you could provide to reproduce this issue? (e.g. does it work when iPhone is reachable from Apple Watch but not when Apple Watch is directly connected to LTE? Or are you using Wifi when not iPhone is not reachable?) What is the resolution of the camera? 

Thanks,
Gaston

DeastinY, 2020-03-15:
Apple Wath Series 5: watchOS 6.1.3
iPhone XR: iOS 13.3.1
Camera is an Logitech C920
OctoPi (Version 1.4.0) is on a Raspberry Pi 4
OctoPod Plugin (0.2.4)

I did not find any setup where it works. App in foreground, reachable, connected to wifi.
The camera works within the app without issues.
Would gladly try any other ideas, thanks for looking into this!

gdombiak, 2020-03-15:
Thanks for sharing all those details. Are you running the camera at 1920 x 1080 pixels? If the answer is yes then I would try reducing the resolution to see if that is the cause so I know what to fix. :)

Thanks,
Gaston

DeastinY, 2020-03-16:
it's running at 720p 15fps

gdombiak, 2020-03-16:
That's great to hear! I guess it has to do with picture size and maybe timing out. I do not have a way to reproduce this unless I purchase a 1080p camera. :(  

Does it work if you use 720p at a fps greater than 15? My guess is that it will. BTW, were you seeing any lag on the phone at 1080p and a higher 15fps? My guess is that the phone was slowly running behind from what was actually happening on the printer due to so much image processing. Now that OctoPrint supports H264, I want to see how/if we can leverage that rather than mjpeg.

Thanks,
Gaston

DeastinY, 2020-03-17:
octoprint support h264 :scream: 
rbr need to check that!

koenkooi, 2020-03-17:
The h264 support is for encoding timelapes only, for cameras it takes whatever mjpeg-streamer puts out, octoprint itself doesn't do anything to that stream.

gdombiak, 2020-03-17:
thanks @koenkooi for clarifying that.

CommanderDer, 2021-11-11:
I have the same problem. At my Iphone X (15.1) is all fine. Camera work fine. At my Apple Watch 7 (8.1) the Camera is not showing. What can i try?