gdombiak/OctoPod #272: Heating temperatures

ScharV, 2020-03-19:
Could you please add the possibillity to change deferred temperatures of Bed and Nozzle? Would be nice for preheating with custom temperatures.

gdombiak, 2020-03-19:
HI @ScharV, have you tried clicking on the pencil to set the temperature? See the image below for an example:
[image]

Or are you looking for something different?
Thanks,
Gaston

gdombiak, 2020-03-19:
You can enter any desired temperature on the "white entry field" at the top right

ScharV, 2020-03-20:
Hi @gdombiak thank you for your fast reply.
Yeah i know this way. I mean it would be lovely to set the most used temperatures.
Hope you understand me🙈
I marked it on your picture.

[image]

I would line to Change the 210 degrees in 200 degrees

gdombiak, 2020-03-21:
Gotcha. This is a nice idea. I will make it so that OctoPod automatically reads temperatures set in OctoPrint's settings:

[image]

Scheduling it for the next release

ScharV, 2020-03-21:
@gdombiak Yeah thanks Sounds Great 💪🏼😬

commit 8b802a14 (2020-03-22): Use OctoPrint's settings for temperatures

Closes #272
closed 2020-03-22 by gdombiak