gdombiak/OctoPod #274: Filamentmanager

ScharV, 2020-03-21:
Hey @gdombiak it‘s me again🙈 Since a few days I‘m using the Filamentmanager Pluging.
https://plugins.octoprint.org/plugins/filamentmanager/
Would ist be possible to implement it to OctoPod? It would be nice only to change the Filament which is presettet on Octopi.
Best regards✌🏼

gdombiak, 2020-03-21:
Thanks @ScharV for this great idea. I like it in particular that this is about selecting the current spool for an extruder/tool. I checked the plugin and it has the required API so this should be feasible. Will add it to the next release since I'm trying to wrap up 3.2 that comes with Apple TV support.

ScharV, 2020-03-21:
@gdombiak thanks for your fast reply, Sounds very good 😬

commit d75d511c (2020-04-26): Added support for FilamentManager plugin. Control your spools from the app. Not implemented: Dialog when new print starts

Closes #274
closed 2020-04-26 by gdombiak