gdombiak/OctoPod #277: Feature request. - enclosure temperature probe

Sarithz, 2020-03-24:
Love the app and what you have done.  One request I’d like to see is the ability to show temperature from a probe such as the DS18B20.  I tried looking at the settings but did not see anything obvious if it is already there.   Once again, wonderful work!

gdombiak, 2020-03-24:
Thanks @Sarithz for the nice words! Nice sensor. I have the same for my other hobbies:
[image]

OctoPod takes all its information from OctoPrint. Is there an OctoPrint plugin that reads from that sensor? I can take a look and see if I can add support for that plugin (if one exists).

Thanks,
Gaston

Sarithz, 2020-03-24:
Yes, sir.  Thanks for the quick response.  I use the Enclosure plugin for OctoPi https://github.com/vitormhenrique/OctoPrint-Enclosure from the plugin repository.

gdombiak, 2020-03-24:
Gotcha. This seems to be a popular plugin so will add support for it in the next release. 3.2 is almost out the door now. I think I will start but showing sensors information. Not sure about other operations. Would that work for you?

Sarithz, 2020-03-24:
Absolutely.  Thanks again if you need any help, let me know.  I’m not an android developer, but am an Apple one. 

Thank you,

James Barrett

AnakinSpain, 2020-04-08:
Now I have 3.2 but I couldn't see the enclosure temperature. Is not implemented still?

gdombiak, 2020-04-08:
Correct. This feature couldn't make it to the 3.2 release and is scheduled for the next 3.3 release. I'm now finishing the work for #286 and then planning to work on this feature.

AnakinSpain, 2020-04-08:
Amazing work!!! Regards!

gdombiak, 2020-04-12:
Progress so far
[image]

AnakinSpain, 2020-04-12:
👍

spatcho, 2020-04-13:
Hi, Great job and looking forward to test it - I m interested by the outputs, do you plan controlling them in that screen or in a new page like you do i.e. with the psu plugin ? I would prefer that later approach.

gdombiak, 2020-04-13:
Hi @spatcho, for the next release I was planning to only expose temp data from this plugin. Now that you asked for output control ... I'm going to look into that and see if I can add it in this release as well. Yes, for output it will appear in its own subpanel as it works for the other plugins.

Thanks,
Gaston

spatcho, 2020-04-14:
Thanks a lot, looking forward, and I will see to play with inputs too :-)

gdombiak, 2020-04-16:
Added support for controlling GPIO and PWM outputs

[image]

closed 2020-04-16 by gdombiak

spatcho, 2020-04-19:
I tested it on the simulator, I love it, thanks a lot ! I use it to turn on/off the printer and I would suggest a "are you sure to turn off" in a further release - usually I m not fan of this but it's easy to miss tap.

gdombiak, 2020-04-19:
Good point @spatcho and thanks for testing out the new feature. Would you mind opening a new ticket for adding a confirmation dialog? I can add a new prompt setting to enable/disable those prompts and also apply your idea to the PSU Control plugin, TPLink/Tasmota/Wemos/etc and Enclosure plugins.

FYI, app was submitted to the store. It will become available to the public once Apple approved it.

Thanks,
Gaston

spatcho, 2020-04-20:
Hi Gaston, thanks, yep i m gonna create the ticket

commit bbeb83c3 (2020-04-16): Added support for controlling GPIO and PWM outputs of Enclosure controller

Closes #277