gdombiak/OctoPod #278: App needs fresh/modern Icon

Matthias1987, 2020-03-28:
The App ist fantastic but the Icon with the blue frog is odd. The App really needs a fresh and modern Icon. Feel free to contact me for any Icon-ideas :)

gdombiak, 2020-04-03:
Hi @Matthias1987, your timing is perfect and thanks so much for helping with this. It is super needed. I'm at work now but will get back to you later today.

Thanks again,
Gaston

Matthias1987, 2020-04-03:
Hi Gaston :)

Thanks for your Mail.
The App-Icon isn’t the final Version. I’ll work on it to look more like an octopus.
But I think this would fit with you App, which I use every day on iPhone, iPad an the Apple TV.

Best Regards
Matthias Petrat
(Editor | Author )

Blog: matthias-petrat.com <https://www.matthias-petrat.com/>
Podcast: Kolumnen-Podcast <https://itunes.apple.com/de/podcast/matthias-petrat-com-kolumnen-podcast/id1030896274?mt=2>
Twitter: @mp_com_blog <https://twitter.com/mp_com_blog>
Facebook: @matthiaspetrat@me.com.blog <https://www.facebook.com/matthiaspetrat.blog/>
Apple Bücher: Apple HomeKit - die Schnittstelle zum Zuhause <https://apple.co/2QOkUEf>

Matthias1987, 2020-04-03:
Just had this idea 💡

Neiljclements, 2020-04-03:
I like where it’s going, but it looks a bit flat and lacking colour / brightness

Matthias1987, 2020-04-04:
Yes, but Icons are flat these days. Changed it a bit.
What do you think!

Neiljclements, 2020-04-04:
I’m probably being dumb - highly likely - but I  can’t see the latest version. Is there a direct link?

Matthias1987, 2020-04-04:
https://www.dropbox.com/s/o2cm5nwea58vxlf/Foto%2004.04.20%2C%2012%2014%2046.jpg?dl=0

Try this :)

jollino, 2020-04-04:
Just my two cents here so take it with a grain of salt. :)

I like the shape of the octopus better, but I'm not sure how I feel about the circle and the flat color around it; my instinct is that it feels like overly convoluted (two background colors, with and blue, with a border in between) without adding much.

I've been staring at some of the app icons I have on my iPhone and I wonder how it would look if the octopus itself were placed directly against a slightly gradiented background (or vice versa: a gradiented octopus against a solid background; or even both), obviously with a much stronger contrast. For instance:

[image] [image] [image] [image] [image] [image]

Or even just a strong contrast against the background, without any gradient:
[image] [image] [image] 
[image] [image] [image]

Maybe the background could be made a little quirky by adding a hint of gcode as a watermark, like Instapaper's icon: 
[image]

Again, just my two cents — Gaston knows that I like to throw all sorts of ideas around. :)

Matthias1987, 2020-04-04:
Thanks for the idea!
So what about this?

Neiljclements, 2020-04-04:
I’m liking where it’s going and I agree about a gradient on the border to give it a bit more pop, if you know what I mean?
And a small thing, could the mouth be a bit bigger? More smiley, if you know what I mean?

gdombiak, 2020-04-04:
Looking better with each iteration. I like @Neiljclements's idea of a more smiley face to make it friendlier and explore the idea of gradient/solid backgrounds as @jollino suggested. @Matthias1987 I think that github is not rendering your last version, all we can see is the text of your reply.

Thanks everyone for making OctoPod a better app. So happy to see this community effort

Neiljclements, 2020-04-04:
Yes, concur with @gaston I’m not seeing updated version you describe.

jollino, 2020-04-04:
Same here, maybe replies via email filter out attachments?
@Matthias1987, can you try posting from Github's page? You should have a link to "view on github" at the bottom of the email. Just drag the file into the textbox and it'll upload and link it up for you.

Matthias1987, 2020-04-04:
I think I got it 🤓
[image]

Neiljclements, 2020-04-04:
Ooh, yes, I like that. Much more cheerful! Would a thin border around the edge frame it a bit better? Slightly darker than the Octopus. 
Can always add it and take it away.
Really like where it’s going. Nice work dude. Making it look a simple is the hardest thing to do.

Matthias1987, 2020-04-04:
Like this?
[image]

Neiljclements, 2020-04-04:
I meant a thin black border, just inside the edges of the icon. Although, the more I see that, the more and more I’m liking it. I think it lacks something and I can’t work our what it is. I feel like the octopus isn’t framed.
The other icons all have a gradient background, but that would probably mean reversing the colours and having the bg green and the octopus white, which I don’t think would work.
I’m not helping, am I?  Do you know what I’m clutching for? It just doesn’t pop like the other icons, and  I absolutely love the octopus graphic and think it deserves to jump out a bit.
Sorry, I’m hopeless....

Neiljclements, 2020-04-04:
What about the octopus orange and the background a gradient dark to light blue? I think that would look awesome. Is it something you can easily try?

Matthias1987, 2020-04-04:
This way?
[image]

Matthias1987, 2020-04-04:
Another try.
[image]
[image]

Matthias1987, 2020-04-04:
The best one I think!
[image]
[image]

Matthias1987, 2020-04-04:
Each time I compared them, I really like the turquoise one with the write background. It is just flat, simple, clean and understandable. Like the Colours on the white background from the Photos-App.

Matthias1987, 2020-04-04:
I think that’s it. Clean, fresh, friendly.
[image]
[image]

Matthias1987, 2020-04-04:
Last try. I made it much more friendly 😁
[image]
[image]
[image]

Neiljclements, 2020-04-04:
Yeah, I’m leaning towards the turquoise more and more. I do like that orange on the dark, but it doesn’t hold me like the turquoise. Could you maybe make the background slightly darker to lift the octopus more?
I think you’ve nailed it though with the subtle changes to the octopus’ face. Has character now.
Really nice work dude. I wish I had that talent to bring life to an idea 👍🏽

gdombiak, 2020-04-04:
Liking a lot this last version. The turquoise is very good since there is some of it in the images used in the buttons of the app.

ScharV, 2020-04-04:
Just an idea 💡 what about 3 or 4 color variants to set in the settings of the app. So the user can choose his favourit design by himself. Like the Home+ App

[image]

Neiljclements, 2020-04-04:
That’s a great idea. I’d forgotten about changing icon colour - perhaps use the orange on blue for one of them? 
Poor @Matthias1987 lol. Bet you wish you’d never started this escapade? Mind you, you’ve done an amazing job and really freshened up the logo.
Top work sir!

Matthias1987, 2020-04-05:
Will create few styles over the day - stay tuned :)

jollino, 2020-04-05:
I like the blue/turquoise background too, after all it's an octopus so it makes sense to have water around it!
Maybe a subtle gradient for the octopus itself (say from solid white to a very pale other color, maybe purplish? that's what I think most people associate octopi with) would give it a little extra punch? Something like this — I'm not a graphic designer so apologies in advance :D 
[image]

Neiljclements, 2020-04-05:
I think the background needs to be darker to pop the octopus out. It looks a bit faded.

Matthias1987, 2020-04-05:
> I like the blue/turquoise background too, after all it's an octopus so it makes sense to have water around it!
> Maybe a subtle gradient for the octopus itself (say from solid white to a very pale [...]

[image]

Matthias1987, 2020-04-05:
Made it friendlier than ever. Gave it some shape (shadows in the background). Changed the colour.
[image]
[image]
[image]

Matthias1987, 2020-04-05:
This are my last Ideas. Users can choose one of them.

[image]
[image]
[image]
[image]

Matthias1987, 2020-04-05:
Okay. Forget the post before.
This two styles are the very last ones. One light and one dark version of it.
[image]
[image]

gdombiak, 2020-04-05:
My vote (and wife’s) goes to the latest from @jollino. Whiteish octopus with some minimal color and gradient gives a nice contrast

Matthias1987, 2020-04-05:
[image]
So this one? 👍

Neiljclements, 2020-04-05:
I like the one you’ve shown above as an icon and in the Apple TV. Perfect!
This one if the link works?

https://www.dropbox.com/s/3ap9es71ub7ce1e/IMG_0665.JPG?dl=0

Neiljclements, 2020-04-05:
Yes, agree with @GDombiak - that’s the one! Let’s put an end to this madness 😂

Matthias1987, 2020-04-05:
Okay. So which files do you need exactly to change the Icon for the Version of iOS, iPadOS and tvOS? 😄

gdombiak, 2020-04-05:
All I need is an SVG and I think with that I can generate the tens of icons. I think no alpha or background transparency if I remember correctly

gdombiak, 2020-04-05:
I think the one from @jollino is a little more whiteish? I’m on a phone so have to scroll and remember. Not ideal

Matthias1987, 2020-04-05:
Changed it more to white on the top ans saved it as SVG. Give it a try 🤓
https://www.dropbox.com/s/l2nkkok9yoh0y6u/OctoPod%20Octopus%20App%20Icon%20Matthias%20Petrat.svg?dl=0

Neiljclements, 2020-04-05:
All I can see is a black square with a pair of eyes 😂

Matthias1987, 2020-04-05:
I think my design won't work as SVG. So I can make you the right Covers for each App as JPEG or PNG.

gdombiak, 2020-04-05:
Ok. I can share Apple’s spec for each icon. It’ll have to be png

gdombiak, 2020-04-05:
@Matthias1987 you are right, the SVG has some borders so I guess I cannot use it.

We need icons for [Apple TV that follow this spec](https://developer.apple.com/design/human-interface-guidelines/tvos/icons-and-images/app-icon/#app-icon-sizes) and for [iOS that follow this spec](https://developer.apple.com/design/human-interface-guidelines/ios/icons-and-images/app-icon/).

For Apple TV we need to provide a background and foreground image and be careful with gradients. All this info is in the links I provided.

Thanks for all your help with this!
Gaston

Matthias1987, 2020-04-05:
Here are all Icon. Just let me know if something is missing or incorrect.
Would be nice if you would link me in the protect because of the fresh Icon.
Just "Matthias Petrat (matthias-petrat.com)"

gdombiak, 2020-04-05:
Wow. Super fast. Will update app later today and share results. Coding now the new dashboard to see all printers in the iOS app in a single page

Matthias1987, 2020-04-05:
Nice. And when you do that, please make these Control-Icons a bit bigger and give them more space between each Icon, please 🤓
[image]

Neiljclements, 2020-04-05:
Amazing job @Matthias1987. You’ve really done a lovely job of that icon 😊👍🏽

Matthias1987, 2020-04-05:
Thank you! 🙏🏻 
I’m glad you like it. Hope to push this project forward with my work. And I am no designer. Did all this work with Affinity Photo on my Mac. 😉
I just love good design.

jollino, 2020-04-05:
Great stuff @Matthias1987!
[image]

On the topic of tweaking the buttons, @gdombiak, I was thinking that maybe the "home" buttons are not the easiest to read...
What if those buttons were the same height as the "move" buttons, and maybe had "X", "Y" and "Z" underneath or on the side? Perhaps the "home all" button could actually say "XYZ" to make it clearer — in hindsight, someone who's not familiar with these commands may get very confused by the home icon alone, especially as in most languages the word itself is different (e.g. in Italian it's "azzera", which has nothing to do with houses or homes.)
As always, just my two cents. :)

gdombiak, 2020-04-08:
[image]

Copied that icon from #291 suggested by @jessearmstrong

Neiljclements, 2020-04-08:
Oh wow! That’s awesome! That looks super professional and I love the gold and black. Stunning!

commit 34e97834 (2020-04-15): Updated OctoPod icon.

Closes #291
Closes #278
closed 2020-04-15 by gdombiak