gdombiak/OctoPod #279: Tasmota plugin not showing on v3.2b4

bdwilson, 2020-03-29:
This previous enhancement (https://github.com/gdombiak/OctoPod/commit/7e7aa4faeba11bb922d88ad00ac6277b1f48e62d) worked fine until the recent upgrade. Only thing I see in the plugins section in Octopod app is "Custom Controls"

bdwilson, 2020-03-31:
It’s the IPPlugViewController that isn’t showing. I now have a Custom Controls view where I used to have my Tasmota control. Tasmota plug-in version is 0.8.11, OctoPod plugin 0.2.5, OctoPrint 1.4 stable. This disappeared recently - I believe it was when 1.4 went stable but worked on RC versions.

gdombiak, 2020-03-31:
Hi @bdwilson, thanks for submitting this issue. There are no recent changes in OctoPod in this area so I wonder what is going on. Was this working fine before? I see that Tasmota 0.8.11 was recently released. Were things working fine with previous version of the plugin? I will install that version and try things locally.

gdombiak, 2020-03-31:
Things appeared as expected. Here is how the plugin is configured
[image]

and this is how it appears in the app:
[image]

Could you share a screenshot of how your plugin is configured? Is the "Tasmota plugs" subpanel not appearing or it appears with no content?

The custom control subpanel always appears even if the custom control editor plugin is not installed.

bdwilson, 2020-04-01:
Thank you for confirming it should still work. Here were my settings. 

https://bubba.d.pr/AuP3co

https://bubba.d.pr/lYFN4O

I removed the Plugin & data then reconfigured as per my old settings and it now shows up. I don’t believe me adding a name this time has anything to do with it.

closed 2020-04-01 by bdwilson