gdombiak/OctoPod #284: Files Not Shown

beakertx, 2020-03-31:
Not seeing the files from my Octoprint server.  Never have after using the app for a while.  Running Octoprint 1.4.0, latest iOS Octopod version.  Most files are on the server, not SD.  Also, can’t add files from iCloud.  Setup error on my part?  Seems like most others don’t have an issue.  Otherwise, AWESOME APP!  Note, I can see the files using a competing app. 

[image]
[image]

gdombiak, 2020-03-31:
Hi @beakertx, sorry things are not working as expected. :( Let's see the first issue before moving to the iCloud issue. If I understand correctly, OctoPod has never been able to get the list of files from your OctoPrint instance when it was running previous versions or now that is running 1.4.0. However, it works fine with other iOS apps. Is this correct?

I would like to reproduce this issue so I can fix it. Would it be possible for you to send me the list of files and folders that you have in your OctoPrint instance? I don't think that plugins could be causing this but who knows. Could you also send me the list of plugins that you have installed?

Thanks,
Gaston

beakertx, 2020-03-31:
Thanks for the quick response!  Attached screenshots of the plugins I have running. I do have a few hundred files, so a bit harder to list those.  Not sure if a good way to do it. But if you suspect it’s a file name, I’m fine with purging. 
Also, running OctoPi 0.15.1, but my understanding is there is no need to reinstall with 0.17 as long as OctoPrint is updated, correct?  #novice

gdombiak, 2020-04-01:
Thanks for sharing more info. At this moment I have 3 possibilities:
1. Big number of files produces a timeout. You have few hundreds files
1. Some character in a file name
1. So many plugins and one of them is affecting this operation

I listed them in the order I suspect they are the reason. Do you have all files in the top folder or do you have them grouped by folders? If you have them all at the top level, could you group them by folders to see if if it works? If it works, then could you open each folder? If it works then I'm almost sure it is a timeout issue. If a given folder fails to load then I would suspect some character in a file name of that folder.

Let me know if you can do this experiment so we can narrow down the cause. Thanks for your patience. :)

Gaston

beakertx, 2020-04-01:
Thanks!  Good suggestions.  Already working on transferring the files from top level to a folder.  If that doesn’t work, I’ll just delete them all!  Cheers!

gdombiak, 2020-04-01:
Great. Thanks for your helping narrowing down the problem. Reorg files into folders and let's see how that goes. Do not delete them yet. :)

Thanks,
Gaston

gdombiak, 2020-04-02:
Any news @beakertx? Curious to hear what you learned from this test?

beakertx, 2020-04-03:
Haven’t stopped printing long enough to find out!  Moving that many files stresses the Pi too much and jitters my prints.  Hopefully have some time this evening.

gdombiak, 2020-04-03:
Good call. Do not bother the RPi while it's printing. :D

beakertx, 2020-04-03:
Success!  Looks like it was one of the files (not sure which one) that was timing out files loading in the app. I don’t blame it for choking on 6 GB of gcode files.  Thanks for your quick responses and suggestions.  Really do love the app, especially the MMU interrupt notifications. Trying to minimize those, but that’s my problem!  Cheers.

beakertx, 2020-04-03:
And uploading from iCloud is working now, too.

gdombiak, 2020-04-04:
@beakertx great to hear that things are working fine now. To be clear, files started to appear as you moved them into folders? If so then it seems like files were not appearing due to the high volume of files being displayed at once. 

Thanks,
Gaston

gdombiak, 2020-04-29:
Added a confirmation dialog when there is a timeout or any error fetching files.

closed 2020-04-29 by gdombiak