gdombiak/OctoPod #288: Mac app

jon-daemon, 2020-04-02:
I love the ios app but I'd really want an app for Mac OS.
I use Octoprint on a system with not much resources and it's always painful to load the web UI especially while printing. Octopod works much better thatn the web UI and I'd really love an app for my computer.
Is this something you consider to build?
Could anyone port the code to Mac?

gdombiak, 2020-04-02:
Hi @jon-daemon, this is an interesting idea. Curious to see if someone else has cycles to look into this.

ariforu, 2020-07-09:
@jon-daemon  I also needed an app to quickly monitor my print job from my laptop. I'm writing a small OctoPod for Mac MVP. Will you be willing to test it?

ariforu, 2020-07-09:
@gdombiak please assign it to me.

jon-daemon, 2020-07-10:
@ariforu Yes of course I want to test it. I have an iMac 12,1 (Mid 2011) with High Sierra 10.13.6.

ariforu, 2020-07-11:
Thanks @jon-daemon . There were some APIs that did not compile in 10.13. I added some API checks to make it compatible. Please unzip and try
[OctoPod Macv0.4.zip.zip](https://github.com/gdombiak/OctoPod/files/4906500/OctoPod.Macv0.4.zip.zip)

It looks like this in Catalina (10.15)
[image]

jessearmstrong, 2020-07-11:
Hot damn that’s cool! Can it also do a camera gallery for multiple printers? That would be the cherry on top.

jon-daemon, 2020-07-11:
Thank you for sharing! I will be at home in 10 days and I will give it a go. I will come back for feedback.

jessearmstrong, 2020-07-12:
I tried it out but it's crashing on me any time I try to open the window after filling out the preferences.
`Process:               OctoPod Mac [68422]
Path:                  /private/var/folders/*/OctoPod Macv0.4.app/Contents/MacOS/OctoPod Mac
Identifier:            org.OctoPod.OctoPod-Mac
Version:               1.0 (1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           OctoPod Mac [68422]
User ID:               502

Date/Time:             2020-07-11 22:55:00.206 -0500
OS Version:            Mac OS X 10.15.5 (19F101)
Report Version:        12
Anonymous UUID:        3A624179-701F-2478-0629-93E2858B552D

Sleep/Wake UUID:       74EE04C4-0554-4B2A-A109-BE3A03FE8314

Time Awake Since Boot: 280000 seconds
Time Since Wake:       6900 seconds

System Integrity Protection: disabled

Notes:                 Translocated Process

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [68422]

Application Specific Information:
Performing @selector(statusBarButtonClickedWithSender:) from sender NSStatusBarButton 0x7fe8ebe1a170

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.OctoPod.OctoPod-Mac       	0x000000010b629d61 0x10b5e0000 + 302433
1   org.OctoPod.OctoPod-Mac       	0x000000010b5e35cb 0x10b5e0000 + 13771
2   org.OctoPod.OctoPod-Mac       	0x000000010b5e71e6 0x10b5e0000 + 29158
3   org.OctoPod.OctoPod-Mac       	0x000000010b608cd5 0x10b5e0000 + 167125
4   org.OctoPod.OctoPod-Mac       	0x000000010b5f6006 0x10b5e0000 + 90118
5   org.OctoPod.OctoPod-Mac       	0x000000010b5f473f 0x10b5e0000 + 83775
6   org.OctoPod.OctoPod-Mac       	0x000000010b5f4cd8 0x10b5e0000 + 85208
7   com.apple.AppKit              	0x00007fff33fdea75 -[NSViewController _sendViewDidLoad] + 87
8   com.apple.AppKit              	0x00007fff340393b7 _noteLoadCompletionForObject + 642
9   com.apple.AppKit              	0x00007fff33f40264 -[NSIBObjectData nibInstantiateWithOwner:options:topLevelObjects:] + 1930
10  com.apple.AppKit              	0x00007fff33fc63be -[NSNib _instantiateNibWithExternalNameTable:options:] + 647
11  com.apple.AppKit              	0x00007fff33fc6042 -[NSNib _instantiateWithOwner:options:topLevelObjects:] + 143
12  com.apple.AppKit              	0x00007fff33fc53b4 -[NSViewController loadView] + 345
13  com.apple.AppKit              	0x00007fff33fc50fb -[NSViewController _loadViewIfRequired] + 72
14  com.apple.AppKit              	0x00007fff33fc5078 -[NSViewController view] + 23
15  com.apple.AppKit              	0x00007fff342cf075 -[NSPopover showRelativeToRect:ofView:preferredEdge:] + 264
16  org.OctoPod.OctoPod-Mac       	0x000000010b606ba9 0x10b5e0000 + 158633
17  org.OctoPod.OctoPod-Mac       	0x000000010b606d5b 0x10b5e0000 + 159067
18  org.OctoPod.OctoPod-Mac       	0x000000010b606e1f 0x10b5e0000 + 159263
19  com.apple.AppKit              	0x00007fff341a4fc7 -[NSApplication(NSResponder) sendAction:to:from:] + 299
20  com.apple.AppKit              	0x00007fff341a4e62 -[NSControl sendAction:to:] + 86
21  com.apple.AppKit              	0x00007fff341a0ebb NSControlTrackMouse + 1745
22  com.apple.AppKit              	0x00007fff341a07c2 -[NSCell trackMouse:inRect:ofView:untilMouseUp:] + 130
23  com.apple.AppKit              	0x00007fff341a0681 -[NSButtonCell trackMouse:inRect:ofView:untilMouseUp:] + 691
24  com.apple.AppKit              	0x00007fff346df3ab -[NSStatusBarButtonCell trackMouse:inRect:ofView:untilMouseUp:] + 305
25  com.apple.AppKit              	0x00007fff3419f9fd -[NSControl mouseDown:] + 748
26  com.apple.AppKit              	0x00007fff3419de10 -[NSWindow(NSEventRouting) _handleMouseDownEvent:isDelayedEvent:] + 4914
27  com.apple.AppKit              	0x00007fff34108611 -[NSWindow(NSEventRouting) _reallySendEvent:isDelayedEvent:] + 2612
28  com.apple.AppKit              	0x00007fff341079b9 -[NSWindow(NSEventRouting) sendEvent:] + 349
29  com.apple.AppKit              	0x00007fff346e0c79 -[NSStatusBarWindow sendEvent:] + 347
30  com.apple.AppKit              	0x00007fff34105d44 -[NSApplication(NSEvent) sendEvent:] + 352
31  com.apple.AppKit              	0x00007fff33f52daf -[NSApplication run] + 707
[... 348 lines cut ...]
Foundation                           4K        1 
Kernel Alloc Once                    8K        1 
MALLOC                           224.9M       63 
MALLOC guard page                   32K        8 
MALLOC_NANO (reserved)           384.0M        1         reserved VM address space (unallocated)
Memory Tag 242                      12K        1 
SQLite page cache                   64K        1 
STACK GUARD                       56.0M        7 
Stack                             11.0M        7 
VM_ALLOCATE                        108K       15 
__DATA                            24.2M      277 
__DATA_CONST                       239K       11 
__FONT_DATA                          4K        1 
__LINKEDIT                       388.5M        5 
__OBJC_RO                         32.2M        1 
__OBJC_RW                         1892K        2 
__TEXT                           170.3M      276 
__UNICODE                          564K        1 
mapped file                       66.1M       25 
shared memory                      644K       16 
===========                     =======  ======= 
TOTAL                              1.3G      756 
TOTAL, minus reserved VM space   985.0M      756 

`

ariforu, 2020-07-12:
Thanks @jessearmstrong for the crash log. Looks like it crashed due to an error in the starscream library that OctoPod uses for websocket connection. I need to look into the starscream library little bit deep but my hunch is that the crash might have happened during hostname parsing. Could you ensure to add http:// before the hostname and host name is valid http URL(not malformed in any way)? Like instead of entering `octopi.local` try entering `http://octopi.local` Start the app fresh and instead of click the status bar icon, click the app icon in the dock and then from the OctoPod Mac main menu select Preferences. 

[image]

jessearmstrong, 2020-07-12:
That did the trick! So far it's working great. Are you planning on supporting multiple printers too? I have 6 that I run almost constantly.

ariforu, 2020-07-13:
Hi @jessearmstrong great. Thanks for confirming. I'll add a URL validation to prevent this from crashing. Supporting multiple printers will be cool. I think it will be beneficial for many. Let's spend some time stabilizing the current app and then we can get started with multi-printer support.

ariforu, 2020-07-13:
@jon-daemon @jessearmstrong Please test the attached(v0.5). Added a URL validation in the preferences dialog. This will prevent the app from saving malformed or protocolless URL there by keeping it from crashed during connection. 

[OctoPod Macv0.5.zip](https://github.com/gdombiak/OctoPod/files/4910590/OctoPod.Macv0.5.zip)

gdombiak, 2020-07-13:
@ariforu this is looking SUPER!!! Many people requested this and now you made it!!

Some quick comments from a quick test:
1. Preferences: Enable command keys. I would use command-A and command-V
1. Camera not showing. I have mutlicam plugin installed but that should not be an issue afaik
1. Connecting status was not rendered. It jumped straight to operational
1. New feature: Use iCloud so printers information is in sync. We need same bundle id to make this happen

ariforu, 2020-07-13:
@gdombiak thanks for testing and providing with feedback. 1 should be easy to fix. For 2 please check if you left a "/" after your host name. I'm using the same same camera absolute URL logic. Will check 3. You are right about 4. We both use individual accounts - so we need to work together on that. 5. May be your about dialog go behind other windows?

jessearmstrong, 2020-07-13:
@ariforu New version working with a hostname that doesn't include http:// at the beginning! Score! I'm excited to see this one develop!

ariforu, 2020-07-14:
@gdombiak 

- [x] Preferences: Enable command keys. I would use command-A and command-V (fixed)
- [x] Camera not showing. I have mutlicam plugin installed but that should not be an issue afaik (should work - seems user error)
- [ ] Connecting status was not rendered. It jumped straight to operational 
- [ ] New feature: Use iCloud so printers information is in sync. We need same bundle id to make this happen (deferring to next version) 
- [x] About OctoPod App does not render about information. Right click on icon to see this menu option (should work - please check if it's going behind other windows)

[OctoPod Macv0.7.zip](https://github.com/gdombiak/OctoPod/files/4916937/OctoPod.Macv0.7.zip)

jon-daemon, 2020-07-21:
I just tested it on High Sierra and it works fine. Thank you! I will come back with feedback after I will use it for a while.

ariforu, 2020-07-23:
> I just tested it on High Sierra and it works fine. Thank you! I will come back with feedback after I will use it for a while.

Thanks. Please do. We'll then work with @gdombiak to merge it with main code base. It's quite a change and touched other OctoPod code which are shared with I/TV & WatchOS.

jon-daemon, 2020-08-15:
Just a few thoughts I'd like to see in the future:
Be able to hide the dock icon
Control buttons (move axis, home axis, change temp tool & bed)
Compatibility with a couple plugins I use. [OpiTemp](https://plugins.octoprint.org/plugins/opitemp/) & [Simple Emergency Stop](https://plugins.octoprint.org/plugins/simpleemergencystop/)

Nappyy, 2020-11-18:
Hi there,
I just wanted to ask if there is any progress after Version 0.7?
In my use case I'd need System Commands and Control Buttons to use this in production.

Thank you for your work so far 👍

Nappyy, 2021-03-21:
any progress?