gdombiak/OctoPod #289: Klipper commands

ScharV, 2020-04-02:
Hey @gdombiak i really love your work and the new AppleTV implementation. Can‘t wait to watch via TV. Last week i flashed my board to klipper and everytime i power on i have to restart the klipper Firmware via Octoprint ui. Would ist be possible To implement the macros of Octoklipper plugin to Octopod? Before i flashed to Klipper i only used Octopod to Start prints and to control.

Thanks again for your Great work

gdombiak, 2020-04-02:
Hi @ScharV, thanks for the nice feedback. I do not have Klipper installed but was looking at the Klipper plugin (had no idea there was one) and by looking at the source code it seems like when you click on any of the macros in the UI, what it does behind the scene is send some gcode to Klipper. 

I looked at the [list of official gcode commands in Klipper](https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md) and I see the **RESTART** and ** FIRMWARE_RESTART** gcode commands. As an alternative to solve your need, have you looked at [Custom Control Editor plugin](https://plugins.octoprint.org/plugins/customControl/)? You could create a command that sends the RESTART gcode. OctoPod has support for the [Custom Control Editor plugin](https://plugins.octoprint.org/plugins/customControl/) so all containers & commands will be rendered and you can run them from OctoPod. This way you can restart Klipper from the app.

Would this solve your need?

Thanks,
Gaston

ScharV, 2020-04-02:
Hey @gdombiak thank you for your fast reply. Yeah i know the Plugin but haven‘t used it yet. Will give him a try and will report 🤙🏼

ScharV, 2020-04-04:
Hey @gdombiak, found out that it works through terminal. Have to try some other workarounds to find the best solution.
Ist would still be nice to implement the octoklipper plugin to use some functions like guided bed leveling or specialized macros like emergency stop.

Thanks,
Marv

kenthinson, 2020-04-29:
> Hey @gdombiak, found out that it works through terminal. Have to try some other workarounds to find the best solution.
> Ist would still be nice to implement the octoklipper plugin, only of possible, to use some functions like guided bed leveling or specialized macros like emergency stop.
> 
> Thanks,
> Marv

I just wanted to add if you have the OctoKlipper plugin installed all normal gcodes are passed through to Klipper as if it was Marlin. Emergency stop: M112 Is a standard gcode so no reason to make a Klipper specific issue.

gdombiak, 2020-10-10:
@kenthinson trying to follow this thread and understand your last comment. Is there something that I should add to OctoPod? or adding a custom control button solves this ask?

Thanks,
Gaston