gdombiak/OctoPod #291: Octopod Icon

jessearmstrong, 2020-04-06:
Since you guys were talking about possibly getting a new icon for the app I'd like to submit a couple of ideas myself.

https://www.dropbox.com/s/vpr9z5qqk6ftu88/Octopod-01.jpg?dl=0
https://www.dropbox.com/s/e5i0nqf4v6ke790/Octopod-02.jpg?dl=0

gdombiak, 2020-04-07:
Wow @jessearmstrong. Could you post those icons in #278 so others can comment. I think we have very nice contenders. The first one looks very cool.

Thanks,
Gaston

gdombiak, 2020-04-10:
@jessearmstrong did you design these icons? I posted it in the other thread and someone asked if it would be possible to add some eyes to the octopus. What do you think?

jessearmstrong, 2020-04-10:
I actually hired someone to make these icons (I am a designer also but I haven't had the time to tackle it myself, and this guy's other work has been very good). But I put the request in with him and I'll post a revision when I hear back.

gdombiak, 2020-04-10:
Thanks @jessearmstrong for your help. Let me know what you hear back. Got really good feedback about https://www.dropbox.com/s/vpr9z5qqk6ftu88/Octopod-01.jpg. The idea of adding a face is to make it feel friendlier if that makes sense.

gdombiak, 2020-04-12:
Hey @jessearmstrong, have you heard anything from the designer? I might need to push a quick release to address #296 (see https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/136 for more info). Chances are that I will update icon on 3.4 since many people are hitting that bug.

Thanks,
Gaston

jessearmstrong, 2020-04-12:
I haven't heard back yet but I was a little slow to get the feedback to him. I'll let you know as soon as I do.

jessearmstrong, 2020-04-13:
[image]
Here's the new version!

gdombiak, 2020-04-13:
Sweet!! Not sure how to turn that image into the icon image.

We need it as an icon for [Apple TV that follow this spec](https://developer.apple.com/design/human-interface-guidelines/tvos/icons-and-images/app-icon/#app-icon-sizes) and for [iOS that follow this spec](https://developer.apple.com/design/human-interface-guidelines/ios/icons-and-images/app-icon/). Can the designer provide those PNG? If he gives us an SVG I can try converting it into PNGs. For Apple TV we need an image for the background and an image for the foreground.

Thanks a lot for the help!!
Gaston

jessearmstrong, 2020-04-13:
I'll get the final deliverables.

jessearmstrong, 2020-04-14:
[Octopod.zip](https://github.com/gdombiak/OctoPod/files/4476814/Octopod.zip)

This is the final package of deliverables, he actually included a bit more than I was expecting. If you need help with these let me know. I have a long history as a designer, but I just didn't have the time to tackle this on my own.

gdombiak, 2020-04-15:
@jessearmstrong , I updated iOS and Apple Watch with the new icon. It's looking good. :) However, for Apple TV I need the following to complete this work:

1. An SVG with the octopus alone (no background)
1. An SVG with black background (entire screen black not only the shape like iOS has)
1. An SVG with the octopus and a full black background (entire screen black not only the shape like iOS has)

I guess that if I have the Octopus alone, I can add a black background with Gimp.

Do you think the designer can provide us with this?

gdombiak, 2020-04-15:
[image]

[image]

jessearmstrong, 2020-04-15:
I can probably handle those. Let me have a quick look.

jessearmstrong, 2020-04-15:
[Octopod (Apple TV).zip](https://github.com/gdombiak/OctoPod/files/4479253/Octopod.Apple.TV.zip)
Is this what you need?

gdombiak, 2020-04-15:
EXCELLENT!!! So fast! Love it. I can now resume this work. I thought I had to wait another day! Yay!

gdombiak, 2020-04-15:
Here is the Apple TV version
[image]

One last request, could the designer fix the case sensitive in the name? He wrote Octopod and the app appears as OctoPod. Maybe in the future we may need the new icon with the name below.

Thanks again and this looks amazing! Wife really loves it!

closed 2020-04-15 by gdombiak

jessearmstrong, 2020-04-15:
Wow! That really does look great! I’ll send that note over to him.

jessearmstrong, 2020-04-15:
[OctoPod-2.zip](https://github.com/gdombiak/OctoPod/files/4481042/OctoPod-2.zip)
Here are the revisions with the Correct case for OctoPod.

commit 34e97834 (2020-04-15): Updated OctoPod icon.

Closes #291
Closes #278