gdombiak/OctoPod #292: up/down control buttons inverted

jon-daemon, 2020-04-07:
I noticed that the up and down buttons don't have the same behaviour with octoprint web UI.
Up is moving the bed foraward but in the web UI it gets the bed closer (so the nozzle moves forward).
I think it's a little confusing. Is there any reason for that?

gdombiak, 2020-04-11:
Wow! Thanks so much for reporting this. I'm shocked that this bug has been there since day one and no one reported it before.

Scheduled it for the next release.

jollino, 2020-04-11:
Wow I had never noticed the invert axis checkbox in OctoPrint, but this bug report explains why I always felt "safer" using it in OctoPod than I ever did in OctoPrint... even though I had never noticed that they were reversed. :D

gdombiak, 2020-04-11:
OctoPod was already reading from OctoPrint whether directions on X/Y/Z axes were inverted. The bug was only present on the Y axis. All buttons in all directions now behave like what you do in OctoPrint itself.

commit 7f9b1280 (2020-04-11): Fixed direction of movements on Y axis

Closes #292
closed 2020-04-11 by gdombiak