gdombiak/OctoPod #295: Connection Failed: Service is down or incorrect port

GoodyGizmos, 2020-04-11:
OctoPod was connecting to OctoPrint on my RPi 3B+ for the first ~2 weeks, but now I cannot connect. I get a Connection Failed: Service is down or incorrect port. I've tried to delete and re-add my OctoPi including resetting the OctoPrint API.

Any recommendations for further troubleshooting?

gdombiak, 2020-04-11:
Hi @GoodyGizmos, it looks like a connectivity issue. As you noticed, resetting API or adding again the printer is not going to fix this. :(

Do you know if you made any changes like: updating OctoPrint, updating OctoPi, updating or changing router configuration, etc? If you did then I would start by reverting the change and see if that solves it. The error message appears because OctoPod got a timeout trying to connect to your OctoPrint instance. If you are using your phone then I would check that you are indeed connected to the local Wifi and not using LTE. If you are using LTE then you will need to make sure that your OctoPrint instance is reachable from outside your network with the hostname that you configured.

One simple test to do is to:
1. Go to Terminal tab and click on OctoPrint Web. That will open a safari browser that will try to connect to the same OctoPrint instance. I'd expect to fail too
1. Copy the same hostname that you used when adding the printer and open Safari. Paste the hostname in the URL and see how it works. Just like previous test I expect it will fail

Let me know how it goes.
Gaston

GoodyGizmos, 2020-04-11:
> Do you know if you made any changes like: updating OctoPrint, updating OctoPi, updating or changing router configuration, etc? If you did then I would start by reverting the change and see if that solves it.
I did try to get run **sudo apt-get update** and **sudo apt-get dist-upgrade**. I guess what I don't understand is why this should matter if I can still connect to the OctoPi via a browser.

>If you are using your phone then I would check that you are indeed connected to the local Wifi and not using LTE. If you are using LTE then you will need to make sure that your OctoPrint instance is reachable from outside your network with the hostname that you configured.
> 
I am on the same WiFi including the same SSID, frequency, and Orbi router.

Well, while troubleshooting with these instruction, I removed/added the printer in OctoPod for like a fifth time and it's working well again now. Strange. I'll let you know if I have any further issues, but thank you for your help. ;-)

closed 2020-04-11 by GoodyGizmos

lewis1286, 2022-03-25:
side note, I had this issue and realized my phone was on my local guest network while octopi was on my main local network.. be sure they are on the same network!