gdombiak/OctoPod #296: Layer Display broken with DisplayLayerProgress 1.19.1

ssnyder30, 2020-04-11:
[image]

gdombiak, 2020-04-11:
Sorry guys about this. I'm in conversations with the developer of the DisplayLayerProgress plugin here https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/136. 

BTW, are you using OctoPrint-OctoPod plugin? Depending on what the developer of the plugin answers I might need to require the user of OctoPrint-OctoPod plugin to see layer information. A lot of users will basically lose seeing layer information if we go this way since not everyone uses OctoPrint-OctoPod plugin.

TheLion, 2020-04-11:
> BTW, are you using OctoPrint-OctoPod plugin? 

Yes, I am.

ssnyder30, 2020-04-12:
I am also using Octopod Plugin.

sschwartzman, 2020-04-13:
For all interested, here's a simple workaround until this is fixed:

In settings for "State", set to `[current_layer] / [total_layers]` like so:
[image]

What you end up with in OctoPod:
[image]

...and in OctoPrint:
[image]
(not ideal long-term, but totally fine by me for the time being)

@gdombiak Thank you for continuing to develop and support OctoPod! I use it religiously for my remote printing/monitoring needs. It's wonderful.

gdombiak, 2020-04-13:
Thanks @sschwartzman. You are spot on, that is another workaround this issue. Thanks for sharing!!!

JaySea77, 2020-04-17:
You asked if I wanted to beta test in the AppStore. Please tell me how.

gdombiak, 2020-04-17:
Hey @JaySea77, good to see you here.  You will need to follow these steps:
1. Install plugin https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/download/1.20.0dev/master.zip. This will replace the version that you have with the new version that will be released. This version requires you to go to Settings and enable WebSocket-Communication. I asked the developer of the plugin to have that enabled by default for the final release
1. Install TestFlight app on your iPhone/iPad if you do not have it already installed. This is how Apple let's you install beta versions of applications
1. Open https://testflight.apple.com/join/Pq7JqNCx to accept installing Beta version of OctoPod
1. If you installed OctoPod in your Apple TV then you will need to also install TestFlight on your Apple TV and from there install the beta version of OctoPod

Let me know how it goes and if things are working fine or not.

Thanks,
Gaston

JaySea77, 2020-04-17:
Thanks, the app now shows the layer again. Will try it with a print tomorrow.

gdombiak, 2020-04-17:
Great to hear that @JaySea77. Thanks for helping test the new version and confirming that things are working. Let me know if you ran into something that needs fixing.

Gaston

TheLion, 2020-04-18:
The dev version of DisplayLayerProgress and the OctoPod app from TestFlight, work great!

gdombiak, 2020-04-18:
New version of the plugin is out. Ready to submit new version of OctoPod to the App Store.

closed 2020-04-18 by gdombiak