gdombiak/OctoPod #297: Siri intent or quick action for printer shutdown

Floppy, 2020-04-11:
My printer has my octoprint server inside it, powered by the same PSU, so whenever I want to turn off my printer, I need to shut down octoprint first. I always end up using Octopod to do that, so it would be really useful to be able to trigger that action via Siri, or by a 3d touch menu action.

gdombiak, 2020-04-12:
Hi @Floppy, I follow your idea but need more information about exactly how you turn things off via OctoPod. Are you asking to be able to execute System Commands via Siri?

Thanks,
Gaston

Floppy, 2020-04-12:
Using the "shutdown system" option in Octopod, under the System Commands menu, yes.

gdombiak, 2020-04-14:
Gotcha. I will see what I can do with Siri command and System Commands. Commands are dynamic and need to check how Siri can handle that. Apple recently improved Siri to accept parameters so will see if that can help with this.

commit 5e24db32 (2020-10-25): Added system command execution from Siri. Some translations are still pending

Closes #297
closed 2020-10-25 by gdombiak