gdombiak/OctoPod #298: hostname is case sensitive (at least on an octopi-0.170/octoprint-1.4.0)

SuperTango, 2020-04-13:
I'm not sure if this is a octopod or octoprint problem.  If it's octoprint let me know and I'll file it there.  

If I add my printer's URL with a case that's different than the machine thinks, the octopod client pops up the "Not refreshing" error banner, and the octoprint.log has repeated errors: 

    2020-04-13 11:56:36,579 - tornado.access - WARNING - 403 GET /sockjs/websocket (::ffff:192.168.1.1) 1.56ms

(After much trial and error) I found that the hostname on the machine is "3dprinterpi.example.com" (it's not really example.com, but you get the picture).  In my octopod printer config, i had the hostname set as "http://3dPrinterPi.example.com".  This seems to cause the 403 errors.  If I change the hostname to "http://3dprinterpi.example.com", it works.  

This is a new install of an octopi with version 0.17.0 and octoprint-1.4.0.  I have no experience with any other version of octoprint, so I'm not sure if this is a new issue or not.

SuperTango, 2020-04-13:
I found this issue in tornado: https://github.com/tornadoweb/tornado/issues/610.  Seems like tornado expects hostnames to be normalized (lower cased?) when the tornado API is called.  Again, not sure if this is an octoprint or octopod issue.

gdombiak, 2020-04-13:
Hi @SuperTango, thanks for reporting this issue. I'm sure others are going to run into this same thing.

I guess you are pointing to [this line](https://github.com/gdombiak/OctoPod/blob/eca5de846e043ae4bdbab3989738d2cc66c5d19f/OctoPod/OctoPrint/WebSocketClient.swift#L66) that includes the hostname as a header. Based on your test it does seem like it is a case sensitive problem. Both [HTTP/1.1](https://tools.ietf.org/html/rfc7230#section-3.2) and [HTTP/2](https://tools.ietf.org/html/rfc7540#section-8.1.2) indicate that headers should be case insensitive. The link you shared is from 2012 and it seems that Tornado fixed it? If this is indeed a bug in Tornado, I think that Gina (OctoPrint) will defer to them since OctoPrint should not be impacting this.

TL;DR: According to the specification, clients (like OctoPod) should be able to send uppercase/lowercase/any mix and things should still work. This could be a Tornado issue or maybe some validation done by OctoPrint? I never looked at that part of OctoPrint's code to know what is going on.

Sorry I cannot help more.
Regards,
Gaston

SuperTango, 2020-04-13:
OK.  Thanks.  I filed an issue in octoprint.

gdombiak, 2020-04-17:
Closing since issue was reported and fixed in OctoPrint (this is a bug in Tornado library)

closed 2020-04-17 by gdombiak