gdombiak/OctoPod #302: HLS Video Feed for Octoprint/Octopod - (Investigation/Enhancement)

jalanjarosz, 2020-04-17:
Gaston,

I've opened an Octoprint ticket partially regarding this request:
https://community.octoprint.org/t/alternate-video-streams/18330

Looking for an alternate way to use a Wyze network camera with Octoprint and hence Octopod.

Users can flash a Wyze camera with RTSP support, which then can be used with a Restreamer docker to send the RTSP feed out as an HLS feed with snapshots.  https://datarhei.github.io/restreamer/

HLS iframe example:
`<iframe src="http://odroid.local:8080/player.html" name="restreamer-player" width="800" height="450" scrolling="no" frameborder="0" webkitallowfullscreen="true" mozallowfullscreen="true" allowfullscreen="true"></iframe>`

HLS direct feed, which opens directly on an iPhone
`http://odroid.local:8080/hls/live.stream.m3u8`

Snapshot URL:
http://odroid.local:8080/images/live.jpg

Since Octopod uses the video feed that Octoprint provides, what are your thoughts of being able to integrate a HLS feed into both Octoprint and Octopod?

Thanks!
James

gdombiak, 2020-04-17:
Hi @jalanjarosz, I think this is a cool idea. OctoPrint itself does not care much about video streaming. OctoPi is the one that adds support for mjpeg streaming. I think that maybe the MultiCam plugin may be interested in this improvement. I can see how each camera in that plugin may have a setting that tells what kind of streaming it uses. 

For cameras that use HLS then OctoPod can use proper code to render HLS. Apple TV and iOs will be fine with this. I do not see Apple Watch support for HLS though. That's a bummer, they have audio streaming but I do not see video streaming to your watch. That would have been cool.

Thanks for submitting this nice idea,
Gaston

RyanMorash, 2020-06-16:
OctoPi is adding built in HLS support, see https://github.com/guysoft/OctoPi/issues/649 and https://github.com/guysoft/OctoPi/pull/650

claytono, 2020-11-11:
HLS support is now in the 1.5.0rc1 release.

gdombiak, 2020-11-12:
NICE! That's great to hear. I guess this is will be available in the next OctoPi release (not OctoPrint). I will check it out and see how to add support for this. This could be a good optimization.

Thanks for sharing the news.
Gaston

claytono, 2020-11-13:
My underrstanding is that the latest OctoPi beta has support for the streaming side of HLS (setting up ffmpeg, etc), but you need the 0.15rc to have octoprint be able to configure and display it also.  I set up the ffmpeg side myself for reasons that aren't interesting, and I've got it working with the 0.15rc with no issues so far.

gdombiak, 2020-11-13:
Thanks @claytono ! That makes sense now that I think of.

claytono, 2020-11-13:
If it's helpful, you can skip the Octopi part and test with just Octoprint 0.15 and any HLS test stream.  I just checked and this one plays just find in the 0.15rc release: https://content.jwplatform.com/manifests/yp34SRmf.m3u8

gdombiak, 2020-11-13:
Great suggestion. I will help A LOT during development! Good idea.

Thanks,
Gaston

gdombiak, 2020-11-24:
iOS app has been updated to handle HLS videos. Unfortunately, Apple Watch is not able to do video streaming. I will see if I can extract an image for the HLS feed so we can send it to the Apple Watch and also to iOS 14 widgets. If not both the Apple Watch and iOS 14 widgets will not have an image. I will update Apple TV app to support HLS feed. 

I wonder if OctoPrint will still do snapshots for HLS feeds. If it does not then push notifications will also not have an image. :(

claytono, 2020-11-24:
@gdombiak I noticed Octopi is saving a JPG snapshot to the side of the HLS video.  I wonder if the intent is to have users configure that as the snapshot url and send that?

gdombiak, 2020-11-24:
Thanks for sharing that info @claytono. I suspect your guess is right. :) 

FYI, Apple's SDK for managing HLS is not as friendly as their newer SDK. Getting a snapshot of an .m3u8 stream is not straight forward .... need more  free time to finish this work!

commit ca77fd24 (2020-11-26): Return image from HLS feed to Apple Watch. Apple Watch cannot stream video so we need to rely on sending a single image as we do for MJEPG.

Closes: #302
closed 2020-11-26 by gdombiak