gdombiak/OctoPod #303: iOS app - Show camera in dashboard of printers

gdombiak, 2020-04-17:
Feedback via TestFlight: "Dashboard view is great. Could it include a camera feed? My default preference would be for camera feed with the printer name rather than stats. So maybe an option to hide either by default? And a tile size option? So they could be a column, or two columns of tiles?"

Maybe add a button to switch to cameras or stats? Maybe have a button for each camera that an octoprint has?

ariforu, 2020-06-25:
@gdombiak isn't it done already?

gdombiak, 2020-06-25:
not implemented yet. Super busy at work so not much free time to work on OctoPod lately.

tideline3d, 2021-01-21:
Adding a vote for this one, especially since there’s now HLS video support.  Wouldn’t want multiple mjpeg streams running on one page but HLS should work well.