gdombiak/OctoPod #305: Webcam not functioning / url broken

jlcook83, 2020-04-18:
I am getting an “unsupported URL” message on the webcam display.  The link below the message is “webcam/?action=stream”.  I have verified that I can reach the webcam through safari using the url http://192.168.0.250/webcam/?action=stream
What is missing in the app url is the IP address.  
All other features function, I can access and control the printer through the app, just a broken webcam url....

gdombiak, 2020-04-18:
Hi @jlcook83, the reason for that error is that a / is missing at the beginning.  Here is an example of my config.yaml

> webcam:
    ffmpeg: /usr/bin/avconv
    snapshot: http://127.0.0.1:8080/?action=snapshot
    stream: /webcam/?action=stream

Look at stream and notice that is starts with /. If you are using multicam plugin then the setup should also include a /. Here is another example

>     multicam:
        _config_version: 1
        multicam_profiles:
        -   URL: /webcam/?action=stream
            isButtonEnabled: false
            name: Default
        -   URL: https://myhome.com/webcam2/
            isButtonEnabled: true
            name: Webcam 1
        -   URL: http://192.168.1.10:8081
            isButtonEnabled: true
            name: Webcam 2

Regards,
Gaston

jlcook83, 2020-04-18:
Yes, that was the issue.  

Thank-you for the very quick answer!

closed 2020-04-18 by jlcook83