gdombiak/OctoPod #306: Error for custom script in Octoprint

TheLion, 2020-04-18:
Hi,

I use some custom scripts to set the brighness of the camera, without having to open the CLI.

[image]

Thsese work fine in Octoprint itself, and somewhere in the past (I don't use them very often, they also worked through the OctoPod app, but now they don't anymore

[image]

The commands run a .sh script in the Pi home-folder.

The "Start video stream" and "Stop video stream" commands also run a .sh scripts in the Pi home-folder, and strangely enough, they do work.

gdombiak, 2020-04-19:
This is interesting @TheLion. There are no recent changes made to system commands so not sure why it stopped working. Have you recently changed something else like updating OctoPrint?

BTW, if you can share your setup (e.g. config.yaml or how to add those system commands) I can try reproducing it and see what is going on.

Thanks,
Gaston

TheLion, 2020-04-21:
> This is interesting @TheLion. There are no recent changes made to system commands so not sure why it stopped working. Have you recently changed something else like updating OctoPrint?

I did update OctoPrint to v1.40 a couple of weeks after it was released, but I'm not sure if that was the moment it stopped working. 

> BTW, if you can share your setup (e.g. config.yaml or how to add those system commands) I can try reproducing it and see what is going on.

I will share the part of config.yaml tomorrow,so hopefully you can reproduce it.

gdombiak, 2020-04-22:
Thanks @TheLion. Once you shared the info I will see of reproducing this issue.

TheLion, 2020-04-22:
The relevant part of the config is:

```
system:
    actions:
    -   action: streamon
        command: /home/pi/scripts/webcam start
        confirm: false
        name: Start video stream
    -   action: streamoff
        command: sudo /home/pi/scripts/webcam stop
        confirm: false
        name: Stop video stream
    -   action: setwebcam default
        command: sudo /home/pi/setwebcam.sh
        confirm: false
        name: Set Webcam Default
    -   action: setwebcam Exposure 800
        command: sudo /home/pi/setwebcam800.sh
        confirm: false
        name: Set Webcam Exposure 800
    -   action: setwebcam Exposure 400
        command: sudo /home/pi/setwebcam400.sh
        confirm: false
        name: Set Webcam Exposure 400
    -   action: setwebcam Exposure 200
        command: sudo /home/pi/setwebcam200.sh
        confirm: false
        name: Set Webcam Exposure 200
```

the .sh files contain, or something similar:

```
v4l2-ctl --set-ctrl=focus_auto=0 --verbose
v4l2-ctl --set-ctrl=focus_absolute=50 --verbose
v4l2-ctl --set-ctrl=exposure_auto=1 --verbose
v4l2-ctl --set-ctrl=exposure_auto_priority=0 --verbose
v4l2-ctl --set-ctrl=exposure_absolute=400 --verbose
v4l2-ctl --set-ctrl=sharpness=200 --verbose
v4l2-ctl --set-ctrl=white_balance_temperature_auto=0 --verbose
v4l2-ctl --set-ctrl=white_balance_temperature=6000 --verbose
```
If you need any more info, let me know.

gdombiak, 2020-04-24:
I think this is enough for me to try reproducing this. Could you do one more thing? Could you go to OctoPrint -> Settings -> Logging and set **tornado.access** to INFO? This will log HTTP traffic to octoprint.log. You are going to see things like

> 2020-04-24 23:09:42,220 - tornado.access - INFO - 204 POST /api/system/commands/core/restart (2601:1c0:4600:1b3:a4d7:1f10:6498:5956) 54.18ms
2020-04-24 23:09:42,228 - tornado.access - INFO - 200 POST /api/system/commands/core/restart (2601:1c0:4600:1b3:a4d7:1f10:6498:5956) 61.93ms

Once you configured this, could you run the system command from OctoPod and then look at octoprint.log and paste here the entries that were logged? I expect OctoPrint to not be returning 200 but some other error code.

Thanks,
Gaston

TheLion, 2020-04-25:
Sure, no problem.

I run the scripts 3 times, once from Octoprint web interface from my Windows computer (192.168.xxx.1), once from OctoPod app on my iPhone outside the network (217.122.82.6) and once from OctoPod on my iPhone inside the network (192.168.xxx.207). External and interal didn't make a difference though. But the requests from the OctoPod app didn't even appear to come through.

From Octoprint webgui
`
2020-04-25 11:31:16,457 - tornado.access - INFO - 200 PUT /plugin/logging/setup/levels (::ffff:192.168.206.1) 18.42ms
2020-04-25 11:31:16,458 - tornado.access - INFO - 200 PUT /plugin/logging/setup/levels (::ffff:192.168.206.1) 19.69ms
2020-04-25 11:31:17,936 - tornado.access - INFO - 200 POST /api/settings (::ffff:192.168.206.1) 1474.15ms
2020-04-25 11:31:17,938 - tornado.access - INFO - 200 POST /api/settings (::ffff:192.168.206.1) 1475.49ms
2020-04-25 11:31:17,999 - tornado.access - INFO - 200 GET /api/system/commands (::ffff:192.168.206.1) 10.94ms
2020-04-25 11:31:18,207 - tornado.access - INFO - 200 GET /api/settings (::ffff:192.168.206.1) 205.33ms
2020-04-25 11:31:18,254 - tornado.access - INFO - 200 GET /api/connection (::ffff:192.168.206.1) 43.53ms
2020-04-25 11:31:18,264 - tornado.access - INFO - 200 GET /api/printer/command/custom (::ffff:192.168.206.1) 8.22ms
2020-04-25 11:31:18,275 - tornado.access - INFO - 200 GET /api/slicing (::ffff:192.168.206.1) 7.81ms
2020-04-25 11:31:18,285 - tornado.access - INFO - 200 GET /api/printer/command/custom (::ffff:192.168.206.1) 8.18ms
2020-04-25 11:31:44,013 - octoprint.server.api.system - INFO - Performing command for custom:setwebcam Exposure 800
2020-04-25 11:31:44,240 - tornado.access - INFO - 204 POST /api/system/commands/custom/setwebcam%20Exposure%20800 (::ffff:192.168.206.1) 246.25ms
2020-04-25 11:31:44,242 - tornado.access - INFO - 200 POST /api/system/commands/custom/setwebcam%20Exposure%20800 (::ffff:192.168.206.1) 248.35ms
`

From outside the network:

`
2020-04-25 11:34:01,571 - tornado.access - INFO - 200 GET /api/job (::ffff:192.168.206.224) 27.36ms
2020-04-25 11:34:01,603 - tornado.access - WARNING - 409 GET /api/printer?exclude=sd,history (::ffff:192.168.206.224) 11.54ms
2020-04-25 11:34:14,512 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:217.122.82.6
2020-04-25 11:35:01,568 - tornado.access - INFO - 200 GET /api/job (::ffff:192.168.206.224) 19.64ms
2020-04-25 11:35:01,601 - tornado.access - WARNING - 409 GET /api/printer?exclude=sd,history (::ffff:192.168.206.224) 9.31ms
2020-04-25 11:35:04,673 - tornado.access - INFO - 101 GET /sockjs/websocket (::ffff:217.122.82.6) 3.74ms
2020-04-25 11:35:04,678 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:217.122.82.6
2020-04-25 11:35:04,937 - tornado.access - INFO - 200 GET /api/version (::ffff:217.122.82.6) 5.37ms
2020-04-25 11:35:04,983 - tornado.access - INFO - 200 GET /api/connection (::ffff:217.122.82.6) 43.85ms
2020-04-25 11:35:04,991 - tornado.access - WARNING - 409 GET /api/printer (::ffff:217.122.82.6) 5.26ms
2020-04-25 11:35:05,102 - tornado.access - INFO - 304 GET /api/settings (::ffff:217.122.82.6) 108.48ms
2020-04-25 11:35:05,127 - tornado.access - INFO - 304 GET /api/printerprofiles (::ffff:217.122.82.6) 20.45ms
2020-04-25 11:35:05,135 - octoprint.server.util.flask - INFO - Passively logging in user _api from ::ffff:217.122.82.6
2020-04-25 11:35:05,137 - octoprint.access.users - INFO - Logged in user: _api
2020-04-25 11:35:05,149 - tornado.access - INFO - 200 POST /api/login (::ffff:217.122.82.6) 44.69ms
2020-04-25 11:35:05,150 - tornado.access - INFO - 200 POST /api/login (::ffff:217.122.82.6) 45.65ms
2020-04-25 11:35:05,154 - octoprint.server.util.sockjs - INFO - User _api logged in on the socket from client ::ffff:217.122.82.6
2020-04-25 11:35:11,840 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:217.122.82.6
`

From inside the network

`
2020-04-25 11:36:40,870 - octoprint.server.util.flask - INFO - Passively logging in user _api from ::ffff:192.168.206.207
2020-04-25 11:36:40,873 - octoprint.access.users - INFO - Logged in user: _api
2020-04-25 11:36:40,878 - tornado.access - INFO - 200 POST /api/login (::ffff:192.168.206.207) 526.85ms
2020-04-25 11:36:40,879 - tornado.access - INFO - 200 POST /api/login (::ffff:192.168.206.207) 527.78ms
2020-04-25 11:36:40,884 - octoprint.server.util.sockjs - INFO - User _api logged in on the socket from client ::ffff:192.168.206.207
2020-04-25 11:36:42,327 - tornado.access - INFO - 200 GET /api/system/commands (::ffff:192.168.206.1) 25.93ms
2020-04-25 11:36:42,562 - tornado.access - INFO - 200 GET /api/settings (::ffff:192.168.206.1) 229.90ms
2020-04-25 11:36:42,610 - tornado.access - INFO - 200 GET /api/connection (::ffff:192.168.206.1) 45.70ms
2020-04-25 11:36:42,621 - tornado.access - INFO - 200 GET /api/slicing (::ffff:192.168.206.1) 8.26ms
2020-04-25 11:36:42,666 - tornado.access - INFO - 200 GET /api/connection (::ffff:192.168.206.207) 42.79ms
2020-04-25 11:36:42,872 - tornado.access - INFO - 200 GET /api/settings (::ffff:192.168.206.207) 203.86ms
2020-04-25 11:36:42,885 - tornado.access - INFO - 200 GET /api/printer/command/custom (::ffff:192.168.206.1) 7.63ms
2020-04-25 11:36:42,896 - tornado.access - INFO - 200 GET /api/printer/command/custom (::ffff:192.168.206.1) 8.32ms
2020-04-25 11:36:42,920 - tornado.access - INFO - 304 GET /api/printerprofiles (::ffff:192.168.206.207) 21.24ms
[... cut ...] cludeRegion", "parameterPattern": "^\\s*(enable|on)(\\s|$)", "description": "Default action to enable exclusion"}, {"action": "disable_exclusion", "type": "AtCommandAction", "command": "ExcludeRegion", "parameterPattern": "^\\s*(disable|off)(\\s|$)", "description": "Default action to disable exclusion"}]}
2020-04-25 11:36:51,899 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.206.207

`
Is this the info you need? Is it the right information? If not, let me know..

gdombiak, 2020-04-25:
Thanks @TheLion. I think that with the info you share I know what is going on. Notice that the commands that fail have an empty space in the action property. To confirm this I just created this command 

>     -  action: dis o bay
>        command: sudo disobay
>        name: Command With Spaces

and got this error

> POST not possible. Invalid URL found. Server: http://192.168.252.17:5000. Service: /api/system/commands/custom/dis o bay

When you run the command from the web UI, the browser is escaping those characters. I will do the same now. In the meantime, replace space with _ like this

>     -  action: dis_o_bay
>        command: sudo disobay
>        name: Command With Spaces

Thanks for helping find the root cause.
Gaston

commit f4c63402 (2020-04-25): Escape action of system commands to be path safe.

Closed #306
closed 2020-04-25 by gdombiak