gdombiak/OctoPod #308: Starscream.WSError

SaJaToGu, 2020-04-21:
I always get this error when opening the cam but I can the the cam without a problem. 
[image]

gdombiak, 2020-04-22:
Hi @SaJaToGu, that error message is not related to the camera but to the websocket. Websocket connection is what the app uses for refreshing data like print time, progress, temps, terminal window, etc. When the app starts it also makes an HTTP request to fetch some data until websocket connection comes active. Having said all this, I'd say you see some data in the app but the data is not getting updated.

Are you using a reverse proxy to connect to OctoPrint?

SaJaToGu, 2020-04-22:
Hi,I do not use a reverse proxy, but printing from the SD card. Is this maybe the problem?
Guido

gdombiak, 2020-04-24:
HI @SaJaToGu, printing from SD card or printing from OctoPrint is unrelated to connectivity problem. BTW, do you have upper case letters in the hostname? There was an issue reported recently about websockets failing to connect due to upper case letters in the URL.

SaJaToGu, 2020-04-25:
Hi,Yes that was the problem.I changed to Ip adress and it is working without any errors now.
Thx
Guido

gdombiak, 2020-04-25:
Great to hear that. This was reported here #298 and was fixed in the next version of OctoPrint. The root problem is in a library used by OctoPrint.

closed 2020-04-25 by gdombiak