gdombiak/OctoPod #310: thumbnails in files

jon-daemon, 2020-04-23:
We are able to have thumbnails in gcode files with PrusaSlicer and they are shown in Octoprint with [this](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails) plugin.
It will be great to see these thumbnails when in file details view

gdombiak, 2020-04-23:
This is such a great idea! Thanks @jon-daemon for filing this issue. Scheduled it for the next release.

gdombiak, 2020-04-25:
[image]
[image]

commit a11bd9e9 (2020-04-25): Added file thumbnails.

Closes #310
Closes #311
closed 2020-04-25 by gdombiak