gdombiak/OctoPod #311: Add support for Ultimaker Format Package plugin

gdombiak, 2020-04-25:
This plugin works just like #310 so we can add support for it with no extra work. Both plugins add the same new field **thumbnail** to the files API

closed 2020-04-25 by gdombiak

jneilliii, 2020-04-25:
@gdombiak this one doesn't yet support the new metadata field, but I was just finishing up that change and going to bed. I will be releasing an update tomorrow for UFP after some more testing. IF anyone wants to get ahead of the curve and do some testing for me, that would be great. Just use the URL below in plugin manager to install.

https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage/archive/0.1.5.zip

gdombiak, 2020-04-25:
Wow. I just created this ticket! :) I looked at the source code and saw the field so I assumed it was in production. My bad. No worries, I will release this code in a week or maybe more (depending on how many things I pack). Take your time to test the plugin! Thanks for all your plugins and hard work!

FYI, I already updated the [wiki page](https://github.com/gdombiak/OctoPod/wiki/Supported-Plugins) with your new plugins.

Regards,
Gaston

commit a11bd9e9 (2020-04-25): Added file thumbnails.

Closes #310
Closes #311