gdombiak/OctoPod #313: Feature Request: Show thumbnail

TheLion, 2020-04-26:
Hi,

It would be really awesome, if OctoPod could show a thumbnail of the print when selecting a file in the "Files" section. 

Take a look at the [PrusaSlicer Thumbnails plugin](https://plugins.octoprint.org/plugins/prusaslicerthumbnails/), maybe you can utilize the functionality from that plugin.

gdombiak, 2020-04-26:
It's already implemented for the next release. :) 

Tickets:
https://github.com/gdombiak/OctoPod/issues/310
https://github.com/gdombiak/OctoPod/issues/311

closed 2020-04-26 by gdombiak

TheLion, 2020-04-26:
> It's already implemented for the next release. :)
> 
> Tickets:
> #310
> #311

Hahaha. Didn't look at the closed issues, didn't think it would be implemented that quick 😜