gdombiak/OctoPod #314: Hide printers from dashboard

taker218, 2020-04-27:
Hi,

I just updated the iOS App to the latest version and the dashboard is pretty nice.
Would it be possible to add an option to the printers, so you can hide them in the dashboard?

In my case I have 4 printers configured, but they are in fact, just 2 printers.
2 configured printers are for when I'm home, so they have the local IP-addresses configured.
The other 2 have the external IP (with http credentials), so I can monitor them from when I'm not at home and so I don't need to activate the VPN everytime I open the app. They don't show the progress or time left in that case, but that's not a big issue for me.

So, TLDR:
It woul be nice to have the option to hide a printer from the dashboard view :)

Best regards,
Thomas

gdombiak, 2020-04-27:
Great idea. Just implemented it and will be available in the next release (coming out this week)

[image]
[image]

Thanks,
Gaston

taker218, 2020-04-27:
awesome, thanks for the quick reply!

commit 4b22be64 (2020-04-28): Printers can now be hidden in dashboard

Closes #314
closed 2020-04-28 by gdombiak