gdombiak/OctoPod #315: Optimize iPad layout

kenthinson, 2020-04-27:
Hello! Thanks for making a great iOS app. I am using it on iPad and it seems like the space is not well used. Like it's the phone app blown up. So I took some screen shots to show the issues I see and the possible fix I have imagined so far. Please take this as constructive suggestions I don't mean any offense. you have done a great job :)

Issue 1 Printer status is hidden
[image]

Issue 2: Scrolling down to see printer status looses file name and Pause / Cancel button
[image]

Lots of empty space that can be used to show things. 
[image]

Here is a idea how to make it better off the top of my head. I'm not sure how the bottom half of the screen space could be used better yet. Would need to think about it more. 

But the top half can show the temp graph and that video feed. It also can move up a little to not loose information off the screen edge. 
[image]

gdombiak, 2020-04-27:
Hi @kenthinson 

Thanks so much for your UI feedback and proposals. You are correct and no offense has been taken. :) Anything to improve the app is more than welcome. You are also correct in that the iPad is the same iPhone UI but in a bigger screen. As you noticed this prevents us from making better usage of the space as you highlighted. 

I will see what I can do to improve UX/UI and yet make it easy to maintain iphone and ipad codebases.

Thanks,
Gaston

gdombiak, 2020-04-29:
This other ticket might be related to this one #318. Both are about improving iPad layout.