gdombiak/OctoPod #316: Allow to slide separator between camera and subpanel

gdombiak, 2020-04-28:
Some people may not have a camera or may not be interested in seeing the camera as big. Allow users to slide the "bar" up to reduce camera screen size. Remember camera size so it can be used again when app starts. Slide down will be limited to what we can see now in the camera. Maybe allow to go bigger as long as image can still render fine.

paviro, 2020-05-11:
I would also really like a possibility to hide the camera as I don't have one. Especially since the App is alerting me about connection errors because of it.

wikolii, 2020-05-22:
I would love to see this implemented since not all my printers have cameras.
Maybe an option in the printers settings?
Thanks!

gdombiak, 2020-06-05:
It is getting implemented! :)

[image]
[image]
[image]

closed 2020-06-06 by gdombiak

wikolii, 2020-06-07:
Hey, nice! Is this getting updated via the apple app store? Great job!

gdombiak, 2020-06-07:
Hey @wikolii, the app has been submitted to the App Store for beta testing. My plan is to push it to the app store for an official release this week. I guess around Wednesday it will be live.

commit 3cb1ecda (2020-06-06): Hide camera subpanel if printer setting says so.

Closes #316