gdombiak/OctoPod #317: Hide Camera Pane if Webcam Support is disabled

schiguoi, 2020-04-28:
It would be nice if I could reclaim some screen space for Panel or Move details for printers that don't have cameras connected to them. I have an Octoprint setup that has webcam support disabled and would prefer to see print stats be the prominent feature when a camera is unavailable.

gdombiak, 2020-04-29:
Another great idea @schiguoi!!! Not sure if it was you that mentioned this in the review of the app in the App Store. Based on the review's feedback I created #316. Do you think that would solve this request too? I was thinking that you can slide the "divider between the camera and the subpanel below" up as a way to hide the camera or decide how much you want to see of the camera. The app will remember your setting for this printer since you might have a camera for other printers.

Thanks,
Gaston

schiguoi, 2020-04-29:
It wasn’t me in the App Store, but I saw that issue after I posted mine here.  I think that solution would be a great one as it would still give the option to show troubleshooting/error details without completely removing the pane.  I like the option of sliding it away and remembering that state better than hiding it completely as I will change my mind later when I get a camera for that octoprint setup.

gdombiak, 2020-04-29:
Are you ok if I close this issue and leave the other one?

Gaston

schiguoi, 2020-04-29:
Works for me!

closed 2020-04-29 by schiguoi