gdombiak/OctoPod #318: iPad Multitasking Support

schiguoi, 2020-04-28:
The current UI is much nicer as an iPhone app today with the way the app scales on a larger screen, and it would be great if I could load as a slideover app or as a split screen app on the iPad. 

It would also be great if I could put two instances of the app side by side to manage two different printers at the same time.  I see this as different than the dashboard feature as the dashboard would likely be an at a glance look where managing two printers might be more of a control center type of setup with an app on the left and another on the right having various prints updated throughout the course of the day.

gdombiak, 2020-04-29:
Thanks @schiguoi for this great idea. I think that this would be an awesome feature. Super convenient way for managing multiple printers with minimal overhead.

gdombiak, 2020-06-06:
Hey @schiguoi, good news. Next version will include split view & slide over support thanks to @andrew-codechimp. Multi instance support is not implemented yet.

schiguoi, 2020-06-06:
Thanks @andrew-codechimp and @gdombiak!