gdombiak/OctoPod #319: App store updated my iPad 4 from 3.2 build 4 release to the latest 3.3, now broken

ericd1949, 2020-04-29:
I thought the update wouldn't happen because the new version does not support iOS 10, but it has.
Is there any way I can get the previous 3.2 release which works fine on iOS 10, re-installed?
The update happened automatically, I did not initiate it.

gdombiak, 2020-04-29:
My bad @ericd1949. Yesterday I realized the same thing. I changed one setting but not another one to indicate that the new version 3.3. can only run on iOS 11 or newer. Sorry about this.

Looking at how to solve this. It seems like I need to release the new version (that requires iOS 11 or newer) and "disable" the 3.3 version. With this you will be able to remove your app and download it again from the app store. But this time the App Store will know to install version 3.2 and not 3.4.

I will update this ticket once the app was released and you can downgrade.

Again so sorry about this.
Gaston

ericd1949, 2020-04-29:
I know very well from experience that stuff does happen, but, sooo very glad it can be fixed. Look forward to it. I was doing some printing this morning (that's how I found it) and without it I realised that I'd forgotten how smoothly Octopod makes the Octoprint printing process, compared to any other user interface.
Cheers.
Eric

gdombiak, 2020-04-29:
Thanks Eric for your understanding. :) 

I submitted the new version to the App Store. I guess it will be live tomorrow. Will keep you posted.

Gaston

ericd1949, 2020-04-30:
Not sure if I've jumped the gun, but yes version 3.4 is now available on App store, but my lowly iOS 10.3.1 is being given v3.3 build 7. Presumably v3.3 hasn't been disabled yet?
Cheers,
Eric

gdombiak, 2020-04-30:
In a meeting .... can you check now and let me know? I disabled the 3.3 version. I guess you need to remove the app and install it again.

ericd1949, 2020-04-30:
It's still loading v3. 3, I've tried deleting and reloading again, in between power recycles and disabling icloud drive, but it still reloads the v3. 3 build 7 as the "last compatible version"

gdombiak, 2020-04-30:
Hmm, not sure what else needs to be done. Here is where we see version 3.3 should not be available for download.

[image]

Maybe the way the app is being download has to do with this? Based on [this](https://apple.stackexchange.com/questions/254172/how-can-i-make-allow-users-with-older-versions-of-ios-to-download-an-older-versi) and [this](https://apple.stackexchange.com/questions/195074/how-do-i-install-an-older-version-of-an-app-on-an-iphone) it seems like to re-download the app you need to go to "My purchases".

Let me know if that helped.
Gaston

ericd1949, 2020-04-30:
Yes indeed, I have now got it, v3. 2 :-),    but it was a bit of a journey.
Deleted 3.3, loaded up again via my purchases, that loaded v3. 1, build 7 (I think),  being experimental, I deleted that version and then reloaded it via the app store, that then loaded v3. 2 build 4.

iOS, you are both mysterious and bizarre.... IMHO  ;-) 

So, thanks very much for your efforts and help Gaston, I'm very happy to have Octopod back again on my aging iPad.
Cheers 
Eric

gdombiak, 2020-04-30:
Happy to hear @ericd1949 that you got it working again with the old version. Thanks for documenting what you did so others that might need this know how to solve it.

Regards,
Gaston

closed 2020-04-30 by gdombiak