gdombiak/OctoPod #320: Enclosure implementation not working for GPIO outputs

LegendaryEngineer, 2020-04-29:
Hi,
I've set up the Enclosure Plugin on my OctoPrint to switch the printer and a light ON/OFF using GPIO's.
When trying to switch these outputs from the App on my Iphone, I always get a warning pop-up that the operation failed.
The log line from the octoprint.log is as following:
_tornado.access - WARNING - 404 GET /plugin/enclosure/outputs/1 (::ffff:10.1.1.15) 15.90ms
tornado.access - WARNING - 404 GET /plugin/enclosure/outputs/2 (::ffff:10.1.1.15) 16.62ms
tornado.access - WARNING - 404 PATCH /plugin/enclosure/outputs/1 (::ffff:10.1.1.15) 19.47ms
tornado.access - INFO - 200 PATCH /plugin/enclosure/outputs/1 (::ffff:10.1.1.15) 22.27ms
tornado.access - WARNING - 404 PATCH /plugin/enclosure/outputs/2 (::ffff:10.1.1.15) 18.96ms
tornado.access - INFO - 200 PATCH /plugin/enclosure/outputs/2 (::ffff:10.1.1.15) 21.91ms
tornado.access - WARNING - 404 PATCH /plugin/enclosure/outputs/1 (::ffff:10.1.1.15) 19.62ms
tornado.access - INFO - 200 PATCH /plugin/enclosure/outputs/1 (::ffff:10.1.1.15) 22.50ms
tornado.access - WARNING - 404 PATCH /plugin/enclosure/outputs/2 (::ffff:10.1.1.15) 18.54ms
tornado.access - INFO - 200 PATCH /plugin/enclosure/outputs/2 (::ffff:10.1.1.15) 21.31ms_

When switching the outputs from the web UI, the following message is logged:
_tornado.access - INFO - 200 GET /plugin/enclosure/setIO?status=false&index_id=1 (::ffff:10.1.1.14) 42.05ms_

Is it possible that the API call from OctoPod is wrong?

Thanks in advance.
BR, Matthias

gdombiak, 2020-04-29:
Hey @LegendaryEngineer, thanks so much for posting the network traffic. That saved me some time troubleshooting this.

I suspect you have an old version of the plugin installed. I'm running Enclosure Plugin (4.13.1). The API call from the web UI is using the old & deprecated API. The one that is returning 404 is the new API that does not exist in your installation. This is why I suspect you have an old version of the plugin installed.

Let me know if that was the case or not.
Thanks,
Gaston

LegendaryEngineer, 2020-04-29:
Hi @gdombiak,
Thanks a lot for your quick feedback.
You were totally right!
I had the Enclosure Plugin version 4.12 installed, because this is the latest version when installing it from the search with the Pluginmanager.
After updating it using the GitHub URL, the version 4.13.1 got installed.
Now it's working like a charm :)

Thanks again for your fast help!
Your App is awesome, many many thanks for your effort!

BR, Matthias

closed 2020-04-29 by LegendaryEngineer