gdombiak/OctoPod #321: Temperatur Not showing correctly in Webcam area (Panel tab)

Nappyy, 2020-05-02:
Hi there,
On iPhone the Webcam area in the Panel tab doesn‘t Shows Temperaturen correctly.
In default State the Temperature isn‘t visible at all:
[image]

If you Try to scroll down, it becomes visible:
[image]

On iPad it Shows well.

But another issue is, That it just shows the T0 temperature, Not T1.
Please add the temperature of the second head also, if installed.

Kind regards,
Kai

gdombiak, 2020-05-02:
HI @Nappyy, this is working as designed. The idea is to show temperature information when user cannot see temperature information on the bottom subpanel. In your first screenshot I see temp is 42 and in the second one temp information is not visible at the bottom so camera show the temp.

About showing temperature of extruders 2-5, I could easily add that information to the camera. My concern is that it might take too much space a block the view of something more important. Do you have some ideas on how to add that information and still not be intrusive and cover camera information?

Thanks,
Gaston

Nappyy, 2020-05-02:
Hi @gdombiak,
ok, I agree to show only the temperature information if needed (not visible at the bottom).
But in my opinion it's necessary to display (in my case) both head temperatures.
Maybe a compromise could be to show only the temperatures of the active components?
In the screenshots above it would be head 2 and heatbed?

Kind Regards,
Kai

gdombiak, 2020-05-03:
I like that idea. My printers have only 1 extruder so not sure how it works with multiple. Do people only heat up one extruder at a time? If people heat up more than one then we can show the first one as we do now?

Thanks for sharing your ideas.
Gaston

gdombiak, 2020-05-03:
I will share a screenshot so we can see how much gets covered if we show up to 5 extruders. That would be a total of 7 rows.....

Nappyy, 2020-05-03:
> I like that idea. My printers have only 1 extruder so not sure how it works with multiple. Do people only heat up one extruder at a time? If people heat up more than one then we can show the first one as we do now?

well, that was a misunderstanding :)
What i meant was to to display all units that have set a target temp.
Sure it will cover space in the webcam view, but in my opinion it's necessary to see all temperature values, not only the first.

If you like to make it more dynamic, maybe add an option to enable / disable the view of each heating unit in the webcam area?

gdombiak, 2020-05-05:
Gotcha. I will see what can be done and still not obstruct camera view.