gdombiak/OctoPod #322: GcodeViewer

Morcegolas, 2020-05-03:
Hello, could gcode viewer be added in the app? I know some gcodes files are big but it could ask like in the webpage.

Thanks.

gdombiak, 2020-05-03:
Hey @Morcegolas, do you mean to see the content of the gcode files or the gcode viewer tab as OctoPrint has? It would be easy to do the former but not the latter. I haven't seen an easy way to access that information or to reproduce the same logic.

Regards,
Gaston

Morcegolas, 2020-05-03:
@gdombiak Thanks for the fast reply, I meant the code viewer tab as OctoPrint, I use it a lot and now I understand why it isn’t in your app.

gdombiak, 2020-05-05:
Gotcha. I will leave the ticket opened in case some day there is enough time and priorities align to make this one happen. Or maybe someone jumps in and contributes this feature.

Thanks for sharing!