gdombiak/OctoPod #323: Manage Timelapses Feature Request

schiguoi, 2020-05-03:
It would be awesome to be able to download my timelapse and octolapse saves from my octoprint setup to my iOS device for quick sharing.

In addition, it would be awesome if I could delete them if I no longer need them on the server.

gdombiak, 2020-05-05:
I like this. Nice idea. Thanks for sharing

ghosilyk, 2021-01-08:
I found my way here to look for this feature request or to find out if it's currently possible. I just want to add my vote for this feature, access to the "Videos and Images..." section would be such an amazing addition to OctoPod. Thanks for the great app.

gdombiak, 2021-01-11:
2 votes. @ghosilyk you just raised the priority for adding this feature. :) Do you have an idea on how it should look in OctoPod? I guess it could be a new subpanel that you see when you swipe to the right/left when in the main Panel.

Thanks,
Gaston

schiguoi, 2021-01-11:
When I suggested it, I was picturing this feature as being something in the Files tab that would allow you to open a folder of videos/images and save them to the camera roll.

I'm not a huge fan of the swiping panels, I always feel like I'm accidentally going to click a button during the swipe trying to get to a specific screen.

gdombiak, 2021-01-12:
@schiguoi , that’s a great idea. I’ll see what I can do. Thanks!!!

Gaston

gdombiak, 2021-01-19:
Making good progress. I can now list timelapse files, delete them, render them and also added support for sharing them.

closed 2021-01-31 by gdombiak

treflip, 2021-01-31:
Awesome. Count me in too... been wanting this feature as well.

mayb also add the ability to start octolapse when printing from octopod too?? I notice if I start a print from the app octolapse fails to start!

gdombiak, 2021-01-31:
hey @treflip , adding support for Octolapse is scheduled for the release after the next one. It is a big effort and I'm short in time these days. Trying to wrap up 3.11 with a few cool features and then onto Octolapse. :)