gdombiak/OctoPod #324: Enclosure plugin doesn’t change gpios

ciapamusche, 2020-05-04:
Hi

When I try to toggle my relays in enclosure plugin in octopod I have error in request.

Other plugins works correctly 

Have some advice?

gdombiak, 2020-05-04:
Hi @ciapamusche, I wonder whether you are running the latest version of the plugin. It should be 4.13.1. This was reported here #320. I will open a ticket with the enclosure developer to address this.

Let me know if that was your case.
Gaston

ciapamusche, 2020-05-04:
Thank you for reply. There is actually a problem with Enclosure plugins. The update you indicated exist and solve the problem but octoprint not recognize automatically the version 4.13.1 So the walkaround is from octoprint page:

Go to settings
Plugin Manager
click Get more.
In the box "From url" paste the master release link
https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/master.zip

This solve the problem.

I hope is useful

gdombiak, 2020-05-04:
Thanks for documenting how to manually solve this until https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/318 is addressed. Closing this ticket for now.

Thanks,
Gaston

closed 2020-05-04 by gdombiak