gdombiak/OctoPod #331: Multiple OctoPrint installations on same Pi cause problems

jopeek, 2020-05-24:
I’m running two installations of Octoprint on the same Pi on separate URLs and while everything works fine in a regular browser, and I am able to add both printers with separate urls into OctoPod, some functionality no longer works. 

I still see the printer feed and temperatures but progress is missing, as are files, etc. An erro is shown in the Panel screen: not refreshing. The operation could not be completed. Starscream.WSError error 1)

Any suggestions?

gdombiak, 2020-06-05:
Hi @jopeek,

I would need to know more about your setup to be able to help troubleshoot this one. Based on what you are sharing, there is a problem opening the websocket connection. Could you share more about your setup? Is this in the local network? Do you have any reverse proxy? What version of OctoPrint you are using? Was this working when you had only OctoPrint instance? How did you configure your Pi to run 2 instances of OctoPrint?

Regards,
Gaston

jopeek, 2020-06-05:
Local network, hardwired, same lan, latest Octoprint. It worked fine before
I switched to dual installs. I just run the entire Octoprint twice on
different ports and then use haproxy so that /cre3 goes to one port and
/mmp2 goes to the other port. I’m happy to share any config files you might
want to see.

Octopod still communicates with the server in some way, for example I am
still getting notifications via the app because I installed the plugin for
that. It’s just the bottom info part of the screens that’s not displaying
anything and showing that exception.

I followed the instructions from this guide and both website versions work
exactly as intended.
http://thomas-messmer.com/index.php/14-free-knowledge/howtos/79-setting-up-octoprint-for-multiple-printers

gdombiak, 2020-06-05:
Hi Jan,

Thanks for sharing that information. I noticed that those instructions for configuring haproxy are missing this:
>  proxy_http_version 1.1;

In [this guide from OctoPrint](https://community.octoprint.org/t/reverse-proxy-configuration-examples/1107) they mention that not having that setting may break websockets. Could you add that setting, restart haproxy and let me know how it goes?

Thanks,
Gaston

gdombiak, 2020-06-05:
oh, that is for Nginx and not haproxy. You see initial data being fetched since HTTP works fine with your haproxy. However, websockets connections are failing to be established so information is not getting refreshed. I suspect that there is something in the haproxy that is not letting your websockets connection pass.

jopeek, 2020-06-05:
You pointed me enough down the right path. If I just add my Octoprint URL
with its port number instead of the haproxy URL to octopod then things work
fine. Thats fine by me :)

Thanks for help.

gdombiak, 2020-06-05:
I think I found the problem! OctoPod does not support _base directories_ and that is what your guide is asking you to do. Instead of accessing each instance at a different port or different hostname, the guide uses /mkc or /anet or /cnc (in their examples) to access to each OctoPrint instance.

So if you write something like http://myserver.com/octo1 to access your instance 1 of OctoPrint and http://myserver.com/octo2 to access your second instance then OctoPod will try opening the websocket to http://myserver.com and of course that will fail.

I'm not sure how risky it would be to change OctoPod to support _base directories_ and not impact existing users. Need to think more about it. In the meantime, of you could provide a different hostname or port and use that to access each instance then it will work. You can also use subdomains if that works for you. For instance, http://octo1.myserver.com and http://octo2.myserver.com. This will work fine in OctoPod.

jopeek, 2020-06-05:
> In the meantime, would it be possible for you to provide a different
> hostname or port and use that to access each instance?. You can also use

Yeah using the different ports works fine :)

gdombiak, 2020-06-05:
I just saw your previous reply. We both arrived to the same conclusion. :) Should we close this ticket?

jopeek, 2020-06-05:
Yeah go ahead and close it. Thank you.

gdombiak, 2020-06-05:
No problem. Let me know if you run into anything else.

Regards,
Gaston

closed 2020-06-05 by gdombiak